prusa3d/Prusa-Firmware-ESP32-Cam #3: Autentification - Set web authentication

sliverbaer, 2024-04-26:
Howdy, just compiled, connected and moved to my IoT network.  On the "Autentification" tab, I tried to set a password, but after setting a password, I would continually get prompted to enter it.  Never seems to take the password I just set.

I then basically have to recompile and upload to the ESP32-CAM and start over.

[image]

Otherwise, so far pretty slick.   Thanks.

to3mmm, 2024-04-28:
This also happened to me. I had to use the factory reset function (GPIO12 to GND for 10 seconds) to get access to the camera again.

johnyHV, 2024-05-03:
Thank you for the report.

This error is peculiar. It looks like a problem with setting the default configuration at the first start of the MCU. I will try to simulate the same problem and find the issue. It may be necessary to add a configuration save check

MarkusReichart, 2024-05-11:
I also experience the this problem.

johnyHV, 2024-05-13:
In the new version of FW 1.0.2-rc1, I added the control of writing data to the EEPROM. In theory, this should help solve the problem

MarkusReichart, 2024-05-14:
> In the new version of FW 1.0.2-rc1, I added the control of writing data to the EEPROM. In theory, this should help solve the problem

Yes, for me this issue is also solved with this version.

closed 2024-05-14 by johnyHV