prusa3d/Prusa-Firmware-ESP32-Cam #4: ESP32 Camera half working

pachek6, 2024-04-27:
Last version of the module installed 1.0.1

The image on connect appears for half second and then just go blank, until next image is loaded and the process is repeated. 90% of the time the image is blank on prusa connect. I tried lower resolution of the capture on the camera to the minimum, to discard it was my internet connection, and the result was equal. 
[image]
[image]

gordonstrodel, 2024-05-01:
I saw this issue when the Camera resolution was 1200px+ and the refresh was 5 seconds. Lowering the resolution to 800px and updating refresh to 10secs eliminated the problem for me @pachek6. Hope that helps!

pachek6, 2024-05-01:
> I saw this issue when the Camera resolution was 1200px+ and the refresh was 5 seconds. Lowering the resolution to 800px and updating refresh to 10secs eliminated the problem for me @pachek6. Hope that helps!

Thanks!!! Will give a try

johnyHV, 2024-05-03:
Thank you for the report.

Prusa connect currently has a limitation of one picture per 10s. But the camera has a minimum limit of 5s. So I have to change the minimum limit in SW.

Can you please try to set the resolution to the maximum size, and check if the problem with incomplete image upload still persists? If you still have a problem with incomplete images, please can you send me the logs from the camera? You can find the logs on the system page, but you need to have a micro SD card inserted

mikromcz, 2024-05-05:
I can confirm all PrusaConnect related issues. I have 30s image, but only a small thumbnail works on Connect, the main, big picture is often incomplete if it's even shown. Usually, it only flashes once and disappears. I will try lowering the resolution as I set the maximum possible.

roqaatch, 2024-05-05:
I have 60s image with max resolution and it works very well

pachek6, 2024-05-09:
I still need to find my  micro SD card to send you the log, however I have observed something funny: even if the image is half loaded, or not loaded at all (grey screen) on the prusa connect screen, when you hit Download Image button the image is totally full. 

**Prusa connect:** 

[image]

**dowloaded image:**

[image]

could not be an issue at the prusa server? the prusa forum and the printables have been lately painfully slow .

johnyHV, 2024-05-09:
@pachek6  If you don't have a micro SD card, there's an alternative method to access the logs via the serial console. You can connect the camera to a PC and utilize software like **Arduino IDE** or **Realterm** to read the output from the camera. The transmission speed is set to **115200 baud 8N1**. It is a more complicated option than getting logs from the camera.

But that's strange because if there were a problem with transmitting the image from the camera, you would only see the top part of the image (because the image is uploaded from top to bottom). But you see the bottom part of the picture

jimwsmith, 2024-05-11:
Same issue on 3 different cameras.
Reduced frequency to 10sec, reduced image size to 640x480 always works.
One Camera works with 1600x1200, other two usually work with 800x600.
Looking at logs the total picture is not uploaded always some bytes short.  SO maybe Prusaconnect interprets this as corrupted image and doesn't show it (or only shows top part of it)

closed 2024-05-17 by johnyHV