prusa3d/Prusa-Firmware-ESP32-Cam #5: no wifi with 1.0.1

yasars, 2024-04-27:
in 1.0 wifi ist working... after update i have no wifi connection only wifi hotspot with 192.168.0.1 is working but no internet :(

go back to 1.0

johnyHV, 2024-05-03:
Thank you for the report.

In FW version 1.0.1, the camera did not see any WiFi networks, or just did not connect to the WiFi network? Please can you send me the logs from the camera for FW 1.0.1 ? You can find the logs on the system page, but you need to have a micro SD card inserted.

yasars, 2024-05-03:
I deleted everything, played fw on it, entered wifi... but after a restart it doesn't connect to the wifi... don't have time for that at the moment, if you put out a new fw and the bug is still there then you'll get it my logs...

yasars, 2024-05-03:
and when I compile it under Arduino, the WiFi server doesn't run either.... the password for connecting the hotspot 12345678 doesn't match...the above happens with your finished image

Fluffybunny96, 2024-05-04:
I have the same issue, how can I provide the logs to help with this ?

johnyHV, 2024-05-05:
Sorry @yasars , but I can't analyze the problem without logs

johnyHV, 2024-05-05:
@Zero-Circle , maybe you have a problem with the driver on the first computer. [here](https://blog.laskakit.cz/wp-content/uploads/2020/03/CH341SER.zip) is the older driver version for CH340.

closed 2024-05-27 by johnyHV