prusa3d/Prusa-Firmware-ESP32-Cam #8: Static IP for Wifi Configuration

BenjaminFalque, 2024-05-01:
Hello ! 
In next version is it possible to add fields/feature to configure wifi with static IP ? 

Thanks a lot,

Regards

Ben

johnyHV, 2024-05-13:
@BenjaminFalque The option to set a static IP address is added in the new release 1.0.2-rc1. You can test it :)

bennyboom38, 2024-05-14:
Hello !  
Thanks for your time to develop this feature!

I have just flashed my espcam with last firmware but I can't see the button advanced wifi setting
Here is the screen shot of the FW version :
[image]

Do I missed something ? 

Thanks for your help

johnyHV, 2024-05-14:
Hello @bennyboom38 it's neccesary clear cache

johnyHV, 2024-05-16:
@BenjaminFalque Did you have time to test the static IP address? It looked functional from my tests

bennyboom38, 2024-05-16:
> @BenjaminFalque Did you have time to test the static IP address? It looked functional from my tests

Hello! 
 yes I was able to configure the static IP, the first time I don't know why I lost the connection after following your procedure ,I had to flash again but then it worked.

So thanks a lot :)

johnyHV, 2024-05-17:
Great. Thank you for the improvement suggestion :) .