prusa3d/Prusa-Firmware-ESP32-Cam #9: Camera Resolution Change only saves on Reboot

gordonstrodel, 2024-05-01:
Firstly, thank you for the excellent tutorial and blog post! :) 
Observation from use: If you want to change the camera resolution via the web portal, you need to hit "reboot" for it to take affect in the configuration. Is this expected behavior?

johnyHV, 2024-05-03:
Thank you. 

No, it's not expected behavior. I tried to simulate this problem, but the resolution always changed without the need to restart the camera. Can you please try it again? Maybe it was accidental, unexpected behavior in your case

Anheledir, 2024-05-03:
For me, the resolution charge was applied without a restart. Just for your information (v1.0.1).

jimwsmith, 2024-05-11:
My resolution changes (after a few seconds)_ and remains - no reboot required

closed 2024-05-13 by johnyHV