prusa3d/Prusa-Firmware-ESP32-Cam #10: Still incomplete image upload with 1.0.1

Anheledir, 2024-05-03:
I flashed the version 1.0.1 and tried different resolution and quality settings to rule out that the image size causes this. While the local website shows the full image, the upload on connect is always incomplete.

**Original photo:**
[image]

**Uploaded photo in connect:**
[image]

johnyHV, 2024-05-03:
Thank you for the report.

Please can you send me the logs from the camera? You can find the logs on the system page, but you need to have a micro SD card inserted. 

I'll check again to see if there might be a specific situation where the software fails to upload the complete image to the backend.

Is this problem still persisting? It could have been a temporary issue with the Prusa Connect backend application.

Anheledir, 2024-05-03:
Sure, see the log below. Couldn't find an error message in it, transmission seems to be ok.

The incomplete photo gets shown on connect for several hours now, doesn't seem to be a temporary issue.

```

----------------------------------------------------------------
Start MCU!
SW Version: 1.0.1 ,Build: Apr 25 2024 11:25:46
Verbose mode: false
Log level: 2
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - CPU reset reason: Reset due to power-on event
0000-00-00_00-00-00 - CPU0 reset reason: POWERON_RESET
0000-00-00_00-00-00 - CPU1 reset reason: EXT_CPU_RESET
0000-00-00_00-00-00 - MCU Temperature: 53.33 *C
0000-00-00_00-00-00 - Internal Total heap: 297472 ,internal Free Heap: 269888
0000-00-00_00-00-00 - SPIRam Total heap: 2095071 ,SPIRam Free Heap: 2068135
0000-00-00_00-00-00 - ChipRevision: 3 ,Cpu Freq: 240 ,SDK Version: v4.4.7-dirty
0000-00-00_00-00-00 - Flash Size: 4194304 ,Flash Speed 80000000
0000-00-00_00-00-00 - PSRAM is used.
0000-00-00_00-00-00 - malloc/new is not using SPIRAM
0000-00-00_00-00-00 - Init cfg module: 371
0000-00-00_00-00-00 - Read FirstMcuStart: 
0000-00-00_00-00-00 - It's not first start MCU: 15
0000-00-00_00-00-00 - web auth enable: 0
0000-00-00_00-00-00 - web auth user: admin
0000-00-00_00-00-00 - web auth pass: ******
0000-00-00_00-00-00 - Load camera CFG from EEPROM
0000-00-00_00-00-00 - Photo quality: 10
0000-00-00_00-00-00 - Framesize: 5
0000-00-00_00-00-00 - Framesize: 5
0000-00-00_00-00-00 - brightness: 0
0000-00-00_00-00-00 - contrast: 0
0000-00-00_00-00-00 - saturation: 0
0000-00-00_00-00-00 - awb: 1
0000-00-00_00-00-00 - awb_gain: 1
0000-00-00_00-00-00 - awb_mode: 3
0000-00-00_00-00-00 - aec2: 1
0000-00-00_00-00-00 - ae_level: 0
0000-00-00_00-00-00 - aec_value: 300
0000-00-00_00-00-00 - gain_ctrl: 1
0000-00-00_00-00-00 - agc_gain: 0
0000-00-00_00-00-00 - bpc: 1
0000-00-00_00-00-00 - wpc: 1
0000-00-00_00-00-00 - raw_gama: 1
0000-00-00_00-00-00 - hmirror: 0
0000-00-00_00-00-00 - vflip: 0
0000-00-00_00-00-00 - lensc: 1
0000-00-00_00-00-00 - exposure_ctrl: 1
0000-00-00_00-00-00 - Camera flash: 1
0000-00-00_00-00-00 - Camera flash time: 150
0000-00-00_00-00-00 - Load PrusaConnect CFG from EEPROM
0000-00-00_00-00-00 - Token: *********************
0000-00-00_00-00-00 - Fingerprint: NjAyMzMxNDEzNDExNTI0OCAzQzpFOTowRTo4Njo3MzpGOA==
0000-00-00_00-00-00 - Refresh interval: 60
0000-00-00_00-00-00 - PrusaConnect hostname: connect.prusa3d.com
0000-00-00_00-00-00 - SSID: FRITZ!Box 7520 WV
0000-00-00_00-00-00 - WiFi password: *********************
0000-00-00_00-00-00 - mDNS: prusa-esp32cam
0000-00-00_00-00-00 - WiFi MAC: 3C:E9:0E:86:73:F8
0000-00-00_00-00-00 - Service AP SSID: ESP32_camera_134.115.248
0000-00-00_00-00-00 - Set WiFi AP mode
0000-00-00_00-00-00 - WiFi STA start
0000-00-00_00-00-00 - WiFi AP start
0000-00-00_00-00-00 - Service IP Address: http://192.168.0.1
0000-00-00_00-00-00 - Scan WI-FI networks
0000-00-00_00-00-00 - Check available WI-FI network: FRITZ!Box 7520 WV
----------------------------------------------------------------
Start MCU!
[... 123 lines cut ...]
2024-05-03_20-35-33 - Photo OK! 0
2024-05-03_20-35-49 - WiFi status: Wifi connected. SSID: FRITZ!Box 7520 WV, BSSID: 0C:72:74:D3:02:81, RSSI: -68 dBm, IP: 192.168.178.71, TX power: 19.5dBm
2024-05-03_20-35-49 - Free RAM: 98140 bytes
2024-05-03_20-35-49 - Free SPIRAM: 1805919 bytes
2024-05-03_20-35-49 - Temperature: 55.56 *C
2024-05-03_20-35-49 - Start sending device information to prusaconnect
2024-05-03_20-35-49 - Data: {"config":{"name":"ESP32-CAM","resolution":{"width":1280,"height":1024},"network_info":{"wifi_mac":"3C:E9:0E:86:73:F8","wifi_ipv4":"192.168.178.71","wifi_ssid":"FRITZ!Box 7520 WV"}}}
2024-05-03_20-35-49 - Sending Info to PrusaConnect
2024-05-03_20-35-50 - Send done: 182 bytes
2024-05-03_20-35-51 - Upload done. Response code: Info: 200 - OK ,BA:Backend available
2024-05-03_20-35-52 - Taking photo...
2024-05-03_20-35-52 - The picture has been saved. Size: 112763 bytes, Photo resolution: 1280 x 1024
2024-05-03_20-35-52 - Photo OK! 0
2024-05-03_20-35-52 - Start sending photo to prusaconnect
2024-05-03_20-35-52 - Sending Photo to PrusaConnect
2024-05-03_20-35-53 - Send done: 80896 bytes
2024-05-03_20-35-54 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-03_20-35-54 - Free RAM: 97908 bytes
2024-05-03_20-36-19 - WiFi status: Wifi connected. SSID: FRITZ!Box 7520 WV, BSSID: 0C:72:74:D3:02:81, RSSI: -69 dBm, IP: 192.168.178.71, TX power: 19.5dBm
2024-05-03_20-36-19 - Free RAM: 97920 bytes
2024-05-03_20-36-19 - Free SPIRAM: 1795871 bytes
2024-05-03_20-36-19 - Temperature: 56.11 *C
```

If you need anything else just tell me how I can support.

Anheledir, 2024-05-03:
I copied a new log with verbose level and retracted only my token, if that helps:

```
2024-05-03_20-43-49 - WiFi status: Wifi connected. SSID: FRITZ!Box 7520 WV, BSSID: 0C:72:74:D3:02:81, RSSI: -70 dBm, IP: 192.168.178.71, TX power: 19.5dBm
2024-05-03_20-43-49 - Free RAM: 100536 bytes
2024-05-03_20-43-49 - Free SPIRAM: 1795871 bytes
2024-05-03_20-43-49 - Temperature: 55.56 *C
2024-05-03_20-43-58 - Taking photo...
2024-05-03_20-43-58 - The picture has been saved. Size: 112402 bytes, Photo resolution: 1280 x 1024
2024-05-03_20-43-58 - Photo OK! 0
2024-05-03_20-43-58 - Start sending photo to prusaconnect
2024-05-03_20-43-58 - Sending Photo to PrusaConnect
2024-05-03_20-43-59 - Send done: 79872 bytes
2024-05-03_20-44-00 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-03_20-44-00 - Free RAM: 100536 bytes
2024-05-03_20-44-19 - WiFi status: Wifi connected. SSID: FRITZ!Box 7520 WV, BSSID: 0C:72:74:D3:02:81, RSSI: -70 dBm, IP: 192.168.178.71, TX power: 19.5dBm
2024-05-03_20-44-19 - Free RAM: 100536 bytes
2024-05-03_20-44-19 - Free SPIRAM: 1795871 bytes
2024-05-03_20-44-19 - Temperature: 55.56 *C
2024-05-03_20-44-28 - WEB server: get json_input
2024-05-03_20-44-28 - {"token":"*****","fingerprint":"NjAyMzMxNDEzNDExNTI0OCAzQzpFOTowRTo4Njo3MzpGOA==","refreshInterval":"60","photoquality":"63","framesize":"5","brightness":"0","contrast":"0","saturation":"0","hmirror":"","vflip":"","lensc":"true","exposure_ctrl":"true","awb":"true","awb_gain":"true","wb_mode":"3","bpc":"true","wpc":"true","raw_gama":"true","aec2":"true","ae_level":0,"aec_value":300,"gain_ctrl":"true","agc_gain":0,"led":"","flash":"true","flash_time":150,"ssid":"FRITZ!Box 7520 WV","bssid":"0C:72:74:D3:02:81","rssi":"-68","rssi_percentage":"64","tx_power":"19.5dBm","ip":"192.168.178.71","wifi_mode":"Client","mdns":"prusa-esp32cam","service_ap_ssid":"ESP32_camera_134.115.248","auth":"","auth_username":"admin","last_upload_status":"Photo: 200 - OK","wifi_network_status":"Connected","log_level":"3","uptime":"0 days, 00:09:47","user_name":"admin","hostname":"connect.prusa3d.com","sw_build":"Apr 25 2024 17:47:50","sw_ver":"1.0.1","sw_new_ver":"v1.0.1"}
2024-05-03_20-44-28 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-28 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-28 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-29 - Read ActifeWifiCfgFlag: 10
2024-05-03_20-44-29 - WiFiWatchdog task. Stack free size: 328 bytes
2024-05-03_20-44-29 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-29 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-29 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-30 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-30 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-30 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-31 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-31 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-31 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-32 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-32 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-32 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-33 - WEB server: /action_capture Take photo
2024-05-03_20-44-33 - Taking photo...
2024-05-03_20-44-33 - The picture has been saved. Size: 112237 bytes, Photo resolution: 1280 x 1024
2024-05-03_20-44-33 - Photo OK! 0
2024-05-03_20-44-33 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-33 - WEB server: Get index.html
2024-05-03_20-44-33 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-33 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-33 - WEB server: Get styles.css
2024-05-03_20-44-33 - WEB server: Get jquery-3.7.0.min.js
2024-05-03_20-44-33 - WEB server: Get scripts.js
2024-05-03_20-44-34 - WEB server: Get esp32_cam.svg
2024-05-03_20-44-34 - WEB server: get photo
2024-05-03_20-44-34 - WEB server: Get refresh-icon.svg
2024-05-03_20-44-34 - WEB server: get json_input
2024-05-03_20-44-34 - {"token":"*****","fingerprint":"NjAyMzMxNDEzNDExNTI0OCAzQzpFOTowRTo4Njo3MzpGOA==","refreshInterval":"60","photoquality":"63","framesize":"5","brightness":"0","contrast":"0","saturation":"0","hmirror":"","vflip":"","lensc":"true","exposure_ctrl":"true","awb":"true","awb_gain":"true","wb_mode":"3","bpc":"true","wpc":"true","raw_gama":"true","aec2":"true","ae_level":0,"aec_value":300,"gain_ctrl":"true","agc_gain":0,"led":"","flash":"true","flash_time":150,"ssid":"FRITZ!Box 7520 WV","bssid":"0C:72:74:D3:02:81","rssi":"-71","rssi_percentage":"58","tx_power":"19.5dBm","ip":"192.168.178.71","wifi_mode":"Client","mdns":"prusa-esp32cam","service_ap_ssid":"ESP32_camera_134.115.248","auth":"","auth_username":"admin","last_upload_status":"Photo: 200 - OK","wifi_network_status":"Connected","log_level":"3","uptime":"0 days, 00:09:52","user_name":"admin","hostname":"connect.prusa3d.com","sw_build":"Apr 25 2024 17:47:50","sw_ver":"1.0.1","sw_new_ver":"v1.0.1"}
2024-05-03_20-44-34 - WEB server: Get page_config.html
2024-05-03_20-44-34 - WEB server: Get light-icon.svg
2024-05-03_20-44-34 - WEB server: Get reboot-icon.svg
2024-05-03_20-44-34 - WEB server: Get jquery-3.7.0.min.js
2024-05-03_20-44-34 - WEB server: Get github-icon.svg
2024-05-03_20-44-34 - WEB server: Get scripts.js
2024-05-03_20-44-34 - WEB server: get json_input
[... cut ...] 0","sw_ver":"1.0.1","sw_new_ver":"v1.0.1"}
2024-05-03_20-44-37 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-37 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-37 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-38 - SerialCfg task. Stack free size: 416 bytes
2024-05-03_20-44-38 - Photo processing task. Stack free size: 6264 bytes
2024-05-03_20-44-38 - System task. Stack free size: 2480 bytes
2024-05-03_20-44-39 - WEB server: Get get_logs.html
```

johnyHV, 2024-05-05:
@Anheledir 

I tried to simulate the same problem, but I couldn't reproduce it. My photo still uploads completely.

```
2024-05-05_21-03-49 - The picture has been saved. Size: 179318 bytes, Photo resolution: 1600 x 1200
2024-05-05_21-03-49 - Photo OK! 0
2024-05-05_21-03-49 - Start sending photo to prusaconnect
2024-05-05_21-03-49 - Sending Photo to PrusaConnect
2024-05-05_21-03-51 - Send done: 179318 bytes
2024-05-05_21-03-52 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available

2024-05-05_21-05-43 - The picture has been saved. Size: 130378 bytes, Photo resolution: 1600 x 1200
2024-05-05_21-05-43 - Photo OK! 0
2024-05-05_21-05-43 - Start sending photo to prusaconnect
2024-05-05_21-05-43 - Sending Photo to PrusaConnect
2024-05-05_21-05-45 - Send done: 130378 bytes
2024-05-05_21-05-46 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
```

May I ask you to delete the FLASH memory in the MCU and upload the firmware to the MCU again? In the first step, it is necessary to click on 'ERASE', and in the second step on 'FLASH'. Unfortunately, this operation deletes the camera configuration, so you will have to set up the camera again.

[image]

I don't understand the problem you are experiencing. Perhaps there is an error in uploading the firmware to the camera.

Anheledir, 2024-05-05:
Running 1.0.1, will erase an reflash tomorrow.

Just asking, I couldn't get that windows tool for flashing running and used this website instead: https://esp.huhn.me
The flashing log showed no errors though, and I flashed several other microcontrollers that way in the past too. But will try the Windows Tool again.

JimBim89, 2024-05-07:
I have a similar issue, but used the normal flash tool. Seems to be related to #4 I guess.

Will investigate further and report back if I have same log issues and if a reflash makes a difference.

johnyHV, 2024-05-09:
@JimBim89 Can you please try how long it takes to ping from your router to connect.prusa3d.com ? Do you have a good wifi signal quality?

jimwsmith, 2024-05-10:
I have same issue with 2 separate cameras - definitely related to file size as if I reduce resolution to 640x480 I get a reliable transmission to Prusa Connect, anything higher and it send only part (or none of the image).  Adjusting image quality seems to make no effect.

jimwsmith, 2024-05-10:
Even at 800x600 the photo size and image size sent don't match
----------------------------------------------------------------
Start MCU!
SW Version: 1.0.1 ,Build: Apr 25 2024 11:25:46
Verbose mode: false
Log level: 2
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - CPU reset reason: Reset due to power-on event
0000-00-00_00-00-00 - CPU0 reset reason: POWERON_RESET
0000-00-00_00-00-00 - CPU1 reset reason: EXT_CPU_RESET
0000-00-00_00-00-00 - MCU Temperature: 76.11 *C
0000-00-00_00-00-00 - Internal Total heap: 297472 ,internal Free Heap: 269888
0000-00-00_00-00-00 - SPIRam Total heap: 4192091 ,SPIRam Free Heap: 4165155
0000-00-00_00-00-00 - ChipRevision: 1 ,Cpu Freq: 240 ,SDK Version: v4.4.7-dirty
0000-00-00_00-00-00 - Flash Size: 4194304 ,Flash Speed 80000000
0000-00-00_00-00-00 - PSRAM is used.
0000-00-00_00-00-00 - malloc/new is not using SPIRAM
0000-00-00_00-00-00 - Init cfg module: 371
0000-00-00_00-00-00 - Read FirstMcuStart: 
0000-00-00_00-00-00 - It's not first start MCU: 15
0000-00-00_00-00-00 - web auth enable: 0
0000-00-00_00-00-00 - web auth user: admin
0000-00-00_00-00-00 - web auth pass: ******
0000-00-00_00-00-00 - Load camera CFG from EEPROM
0000-00-00_00-00-00 - Photo quality: 10
0000-00-00_00-00-00 - Framesize: 3
0000-00-00_00-00-00 - Framesize: 3
0000-00-00_00-00-00 - brightness: 1
0000-00-00_00-00-00 - contrast: 0
0000-00-00_00-00-00 - saturation: 1
0000-00-00_00-00-00 - awb: 1
0000-00-00_00-00-00 - awb_gain: 1
0000-00-00_00-00-00 - awb_mode: 0
0000-00-00_00-00-00 - aec2: 1
0000-00-00_00-00-00 - ae_level: 0
0000-00-00_00-00-00 - aec_value: 570
0000-00-00_00-00-00 - gain_ctrl: 0
0000-00-00_00-00-00 - agc_gain: 15
0000-00-00_00-00-00 - bpc: 1
0000-00-00_00-00-00 - wpc: 1
0000-00-00_00-00-00 - raw_gama: 1
0000-00-00_00-00-00 - hmirror: 0
0000-00-00_00-00-00 - vflip: 0
0000-00-00_00-00-00 - lensc: 1
0000-00-00_00-00-00 - exposure_ctrl: 1
0000-00-00_00-00-00 - Camera flash: 1
0000-00-00_00-00-00 - Camera flash time: 150
0000-00-00_00-00-00 - Load PrusaConnect CFG from EEPROM
0000-00-00_00-00-00 - Token: *********************
0000-00-00_00-00-00 - Fingerprint: ODI0OTIyNDIzNjIzMTc2IDA4OkY5OkUwOkVDOjE3OkIw
0000-00-00_00-00-00 - Refresh interval: 10
0000-00-00_00-00-00 - PrusaConnect hostname: connect.prusa3d.com
0000-00-00_00-00-00 - SSID: AINSLEY MiFi
0000-00-00_00-00-00 - WiFi password: **********
0000-00-00_00-00-00 - mDNS: prusa-esp32cam
0000-00-00_00-00-00 - WiFi MAC: 08:F9:E0:EC:17:B0
0000-00-00_00-00-00 - Service AP SSID: ESP32_camera_236.23.176
0000-00-00_00-00-00 - Set WiFi AP mode
0000-00-00_00-00-00 - WiFi STA start
0000-00-00_00-00-00 - WiFi AP start
0000-00-00_00-00-00 - Service IP Address: http://192.168.0.1
0000-00-00_00-00-00 - Scan WI-FI networks
0000-00-00_00-00-00 - Check available WI-FI network: AINSLEY MiFi
0000-00-00_00-00-00 - WiFi networks scan done
0000-00-00_00-00-00 - 5 networks found
0000-00-00_00-00-00 - Nr | SSID                             | RSSI | CH | BSSID             | Encryption
0000-00-00_00-00-00 -  1 | AINSLEY MiFi                     |  -43 |  6 | B8:6A:F1:2A:A9:62 | WPA2
0000-00-00_00-00-00 -  2 | PLUSNET-XZC33F-Mag               |  -90 |  1 | 4C:22:F3:AA:1D:C9 | WPA2
0000-00-00_00-00-00 -  3 | BT-H5AFFQ                        |  -92 | 11 | 0C:8E:29:3F:EA:40 | WPA2
0000-00-00_00-00-00 -  4 | EE WiFi                          |  -93 | 11 | 62:8E:29:3F:EA:41 | open
[... 205 lines cut ...]
2024-05-10_18-44-22 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-10_18-44-22 - Free RAM: 98044 bytes
2024-05-10_18-44-22 - Taking photo...
2024-05-10_18-44-22 - The picture has been saved. Size: 51477 bytes, Photo resolution: 800 x 600
2024-05-10_18-44-22 - Photo OK! 0
2024-05-10_18-44-27 - WiFi status: Wifi connected. SSID: AINSLEY MiFi, BSSID: B8:6A:F1:2A:A9:62, RSSI: -45 dBm, IP: 192.168.1.70, TX power: 19.5dBm
2024-05-10_18-44-27 - Free RAM: 98044 bytes
2024-05-10_18-44-27 - Free SPIRAM: 4059035 bytes
2024-05-10_18-44-28 - Temperature: 81.67 *C
2024-05-10_18-44-30 - Taking photo...
2024-05-10_18-44-30 - The picture has been saved. Size: 51228 bytes, Photo resolution: 800 x 600
2024-05-10_18-44-30 - Photo OK! 0
2024-05-10_18-44-30 - Start sending photo to prusaconnect
2024-05-10_18-44-30 - Sending Photo to PrusaConnect
2024-05-10_18-44-32 - Send done: 49152 bytes
2024-05-10_18-44-33 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-10_18-44-33 - Free RAM: 98044 bytes
2024-05-10_18-44-41 - Taking photo...
2024-05-10_18-44-41 - The picture has been saved. Size: 49980 bytes, Photo resolution: 800 x 600
2024-05-10_18-44-41 - Photo OK! 0
2024-05-10_18-44-41 - Start sending photo to prusaconnect
2024-05-10_18-44-41 - Sending Photo to PrusaConnect
2024-05-10_18-44-43 - Send done: 49152 bytes
2024-05-10_18-44-44 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-10_18-44-44 - Free RAM: 98044 bytes

jimwsmith, 2024-05-10:
Even stranger - at certain times of the day I can raise the camera resolution and it transmits fine to Prusa connect, other times I have to really reduce it.   wonder if it is a problem with the Prusa Connect server getting overloaded with too may people sharing camera images.

jimwsmith, 2024-05-10:
2024-05-10_19-58-56 - The picture has been saved. Size: 194988 bytes, Photo resolution: 1600 x 1200
2024-05-10_19-58-56 - Photo OK! 0
2024-05-10_19-58-56 - Start sending photo to prusaconnect
2024-05-10_19-58-56 - Sending Photo to PrusaConnect
2024-05-10_19-58-57 - WiFi status: Wifi connected. SSID: AINSLEY MiFi, BSSID: B8:6A:F1:2A:A9:62, RSSI: -41 dBm, IP: 192.168.1.70, TX power: 19.5dBm
2024-05-10_19-58-57 - Free RAM: 50372 bytes
2024-05-10_19-58-58 - Free SPIRAM: 3771035 bytes
2024-05-10_19-58-58 - Temperature: 87.22 *C
2024-05-10_19-58-58 - Send done: 192512 bytes
2024-05-10_19-58-59 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-10_19-58-59 - Free RAM: 100796 bytes
2024-05-10_19-59-07 - Taking photo...
2024-05-10_19-59-07 - The picture has been saved. Size: 195610 bytes, Photo resolution: 1600 x 1200
2024-05-10_19-59-07 - Photo OK! 0
2024-05-10_19-59-07 - Start sending photo to prusaconnect
2024-05-10_19-59-07 - Sending Photo to PrusaConnect
2024-05-10_19-59-09 - Send done: 193536 bytes
2024-05-10_19-59-10 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
2024-05-10_19-59-10 - Free RAM: 100796 bytes

pachek6, 2024-05-10:
> I have same issue with 2 separate cameras - definitely related to file size as if I reduce resolution to 640x480 I get a reliable transmission to Prusa Connect, anything higher and it send only part (or none of the image). Adjusting image quality seems to make no effect.
> 
> Snapshots load full image even at 1600x1200

Just tried 640X480 and now I get no image chopping

jimwsmith, 2024-05-10:
Now people have gone to bed I have one camera on 1600x1200 and the second on 800x600 no chopping (earlier in the day they both had to be reduced to 640x480)

jimwsmith, 2024-05-11:
Even when images are uploaded and display in full it doesn't seem to transmit the same number of bytes as the photo size (see logs above)

johnyHV, 2024-05-13:
Unfortunately, I am unable to simulate this problem. In the newer version 1.0.2-rc1 of the firmware, there are a few other changes. Please try to see if you still encounter the issue with incomplete image uploads. If so, please enable verbose log level and send me the log outputs of at least 10 uploaded images.

jimwsmith, 2024-05-14:
Sadly still same issue - but not always, for a few minutes I did get 1600x1200 to work but now reliably broken.
Attached two log files, one for 1600x1200 (failing, one for 800x600 working)
[800x600SysLog.log](https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/files/15306248/800x600SysLog.log)
[1600x1200SysLog.log](https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/files/15306249/1600x1200SysLog.log)

jimwsmith, 2024-05-14:
Seems to be related to not properly recognising size of photo data.  The photo size is never matching the size reported when photo is first saved but on 1600x1200 resolution it is way off.  The system seems to actually send the number of bytes it thinks is the photo size but this appears to be only part of the image.
1600x1200 fails:
2024-05-14_09-15-56 - The picture has been saved. Size: **210158** bytes, Photo resolution: 1600 x 1200
2024-05-14_09-15-56 - Photo OK! 0
2024-05-14_09-15-56 - Start sending photo to prusaconnect
2024-05-14_09-15-56 - Photo size: **62464**
2024-05-14_09-15-56 - Sending Photo to PrusaConnect, **86016** bytes
2024-05-14_09-15-56 - Connecting to server...
2024-05-14_09-15-56 - SerialCfg task. Stack free size: 716 bytes
2024-05-14_09-15-57 - System task. Stack free size: 2480 bytes
2024-05-14_09-15-57 - Connected to server!
2024-05-14_09-15-57 - Send data photo
2024-05-14_09-15-57 - 86016/0
2024-05-14_09-15-57 - 86016/5000
2024-05-14_09-15-57 - 86016/10000
2024-05-14_09-15-57 - 86016/15000
2024-05-14_09-15-57 - 86016/20000
2024-05-14_09-15-57 - 86016/25000
2024-05-14_09-15-57 - 86016/30000
2024-05-14_09-15-57 - 86016/35000
2024-05-14_09-15-57 - 86016/40000
2024-05-14_09-15-57 - 86016/45000
2024-05-14_09-15-57 - 86016/50000
2024-05-14_09-15-57 - 86016/55000
2024-05-14_09-15-57 - 86016/60000
2024-05-14_09-15-57 - 86016/65000
2024-05-14_09-15-57 - 86016/70000
2024-05-14_09-15-57 - 86016/75000
2024-05-14_09-15-57 - SystemLed task. Stack free size: 404 bytes
2024-05-14_09-15-57 - 86016/80000
2024-05-14_09-15-57 - SerialCfg task. Stack free size: 716 bytes
2024-05-14_09-15-57 - 86016/85000
2024-05-14_09-15-57 - 86016/86016
2024-05-14_09-15-57 - Send done: **86016** bytes

800x600 works:
2024-05-14_09-23-26 - The picture has been saved. Size: **53743** bytes, Photo resolution: 800 x 600
2024-05-14_09-23-26 - Photo OK! 0
2024-05-14_09-23-26 - Start sending photo to prusaconnect
2024-05-14_09-23-26 - Photo size: **54272**
2024-05-14_09-23-26 - Sending Photo to PrusaConnect, **54272** bytes
2024-05-14_09-23-26 - Connecting to server...
2024-05-14_09-23-27 - SerialCfg task. Stack free size: 796 bytes
2024-05-14_09-23-27 - System task. Stack free size: 2480 bytes
2024-05-14_09-23-27 - Connected to server!
2024-05-14_09-23-27 - Send data photo
2024-05-14_09-23-27 - 54272/0
2024-05-14_09-23-27 - 54272/5000
2024-05-14_09-23-27 - 54272/10000
2024-05-14_09-23-27 - 54272/15000
2024-05-14_09-23-27 - 54272/20000
2024-05-14_09-23-27 - 54272/25000
2024-05-14_09-23-27 - 54272/30000
2024-05-14_09-23-27 - 54272/35000
2024-05-14_09-23-27 - 54272/40000
2024-05-14_09-23-28 - 54272/45000
2024-05-14_09-23-28 - 54272/50000
2024-05-14_09-23-28 - 54272/54272
2024-05-14_09-23-28 - Send done: **54272** bytes

jimwsmith, 2024-05-14:
In camera.cpp line 290 you display framebuffer->len as photo size but in line 481 this gets cast to (int)

johnyHV, 2024-05-14:
this is okay. 
int      -> maximum value is 2 147 483 647
size_t -> maximum value ist 4 294 967 295

I don't have same mistake with the photo size. But I see in the backend incompleted photo upload. So this looks as issue with photo upload. Can you send me you mistake  photo from connect ?

jimwsmith, 2024-05-15:
[image]
[image]
Cameras set to 800x600

jimwsmith, 2024-05-15:
[image]
[image]
Cameras set to 1600x1200

jimwsmith, 2024-05-15:
I don't think issue is with the upload as when you look in verbose logs I sent you, the upload successfully sends the (incorrect) image size.

The problem is the image size is corrupted.  after correctly storing the image, then next time we see the size (after (int) cast the size is corrupted.

I will have to try installing Arduino IE as I normally work with VSCode but that refuses to compile the code as it cannot find rtc.h and I don't know what library you used for that.

johnyHV, 2024-05-15:
apparently I discovered the problem and its solution. I still need to test it. but it looks good so far

jimwsmith, 2024-05-15:
> apparently I discovered the problem and its solution. I still need to test it. but it looks good so far

Let me know if you want me to test anything

johnyHV, 2024-05-16:
@jimwsmith I currently added a new release 1.0.2-rc2. Can you tested it please ? I changed the method of sending data, and it should be more reliable. Currently, the buffer where the photo is stored is directly used. So it's a direct approach to photography

Anheledir, 2024-05-17:
> I currently added a new release 1.0.2-rc2. Can you tested it please ?

It now works reliably for me, all shot photos of the last hour were uploaded and visible completely on Prusa Connect.

Thanks to everybody for the additional tests, logs, and eventually the bugfix 🫡👍🏻

jimwsmith, 2024-05-17:
Works well now - three cameras all running at 1600x1200, high quality, 10s refresh

In logs, photo size saved and photo size sent to Prusa connect now match up exactly.

johnyHV, 2024-05-17:
Thank you, @Anheledir and @jimwsmith, for your help in finding the problem and for the time spent testing

JimBim89, 2024-05-17:
Can also confirm that it's working for me now!

closed 2024-05-19 by Anheledir