prusa3d/Prusa-Firmware-ESP32-Cam #12: Successfully flashed firmware, can't tell if the board is booting...

moofie74, 2024-05-05:
After flashing the firmware to the board, I'm unable to see the board in Service AP Mode on the WiFi search screen on my computer.  The instructions allude to the MCU ID, but I'm unable to see any indication of that ID on the MCU board.

It's not obvious if the board has booted:  I see the flash LED flickering faintly, but no other indication other than the bright red LED by the USB power input.

Fluffybunny96, 2024-05-05:
Hi @moofie74, this seems to be the same as #5. Try using another PC as this sorted the issue out for me.

moofie74, 2024-05-05:
> Hi @moofie74, this seems to be the same as #5. Try using another PC as this sorted the issue out for me.

Thanks for that.  Hmm...another PC is going to be a challenge.

johnyHV, 2024-05-05:
@moofie74 Have you erased the FLASH memory before the first FW flash? Alternatively, you might consider trying an older [driver](https://blog.laskakit.cz/wp-content/uploads/2020/03/CH341SER.zip) version for CH340.

moofie74, 2024-05-05:
I did erase the flash first.  I’ll see if an older driver gets me anywhere.  Thank you!

moofie74, 2024-05-05:
@johnyHV I got your other build from [here](https://github.com/johnyHV/PrusaConnect_ESP32-CAM?tab=readme-ov-file) to load through the IDE, and it worked great.  Then I ran this build on my Mac, and that worked fine too.  So, yeah, maybe a weird driver thing with the CH340 on this computer?  But I'm online and it's working like a champ.

Thanks for the awesome software!  This is really cool.

closed 2024-05-05 by moofie74