prusa3d/Prusa-Firmware-ESP32-Cam #13: ESP32 Reset continuously after flashing

Airforc, 2024-05-06:
Flashed my ESP32 cam board (with OV2640 and the correct pinout for the programmer), also set to 'erase all flash prior to upload'.

After flashing board, board service AP becomes somewhat unstable, serial commands do work but MCU appears to be in reboot loop.

CPU reset reason is SW_CPU_RESET, but looks like camera init failed with error 105. I have gotten the camera to show up with an image before, and I have checked the connection of the ribbon cable to the board.

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
----------------------------------------------------------------
Start MCU!
Prusa ESP32-cam https://prusa3d.cz
SW Version: 1.0.1
Build: May  6 2024 16:17:04
----------------------------------------------------------------
Init Logs library
Start init micro-SD Card
E (505) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (506) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
SD Card Mount Failed
Micro-SD card not found! Disable logs
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - CPU reset reason: Software reset via esp_restart
0000-00-00_00-00-00 - CPU0 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - CPU1 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - MCU Temperature: 53.33 *C
0000-00-00_00-00-00 - Internal Total heap: 263880 ,internal Free Heap: 238052
0000-00-00_00-00-00 - SPIRam Total heap: 4192107 ,SPIRam Free Heap: 4191847
0000-00-00_00-00-00 - ChipRevision: 1 ,Cpu Freq: 240 ,SDK Version: v4.4.5
0000-00-00_00-00-00 - Flash Size: 4194304 ,Flash Speed 80000000
0000-00-00_00-00-00 - PSRAM is used.
0000-00-00_00-00-00 - malloc/new is not using SPIRAM
0000-00-00_00-00-00 - Init cfg module: 371
0000-00-00_00-00-00 - Read FirstMcuStart: 
0000-00-00_00-00-00 - It's not first start MCU: 15
0000-00-00_00-00-00 - web auth enable: 0
0000-00-00_00-00-00 - web auth user: admin
0000-00-00_00-00-00 - web auth pass: ******
0000-00-00_00-00-00 - Load camera CFG from EEPROM
0000-00-00_00-00-00 - Photo quality: 10
0000-00-00_00-00-00 - Framesize: 0
0000-00-00_00-00-00 - Framesize: 0
0000-00-00_00-00-00 - brightness: 0
0000-00-00_00-00-00 - contrast: 0
0000-00-00_00-00-00 - saturation: 0
0000-00-00_00-00-00 - awb: 1
0000-00-00_00-00-00 - awb_gain: 1
0000-00-00_00-00-00 - awb_mode: 0
0000-00-00_00-00-00 - aec2: 0
0000-00-00_00-00-00 - ae_level: 0
0000-00-00_00-00-00 - aec_value: 300
0000-00-00_00-00-00 - gain_ctrl: 1
0000-00-00_00-00-00 - agc_gain: 0
0000-00-00_00-00-00 - bpc: 1
0000-00-00_00-00-00 - wpc: 1
0000-00-00_00-00-00 - raw_gama: 1
0000-00-00_00-00-00 - hmirror: 0
0000-00-00_00-00-00 - vflip: 0
0000-00-00_00-00-00 - lensc: 1
0000-00-00_00-00-00 - exposure_ctrl: 1
0000-00-00_00-00-00 - Camera flash: 0
0000-00-00_00-00-00 - Camera flash time: 200
0000-00-00_00-00-00 - Load PrusaConnect CFG from EEPROM
0000-00-00_00-00-00 - Token: *
0000-00-00_00-00-00 - Fingerprint: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
0000-00-00_00-00-00 - Refresh interval: 30
0000-00-00_00-00-00 - PrusaConnect hostname: connect.prusa3d.com
0000-00-00_00-00-00 - SSID: XXX
0000-00-00_00-00-00 - WiFi password: ***********
0000-00-00_00-00-00 - mDNS: prusa-esp32cam
0000-00-00_00-00-00 - WiFi MAC: 24:6F:28:88:30:E8
0000-00-00_00-00-00 - Service AP SSID: ESP32_camera_136.48.232
0000-00-00_00-00-00 - Set WiFi AP mode
0000-00-00_00-00-00 - WiFi STA start
0000-00-00_00-00-00 - WiFi AP start
0000-00-00_00-00-00 - Service IP Address: http://192.168.0.1
0000-00-00_00-00-00 - Scan WI-FI networks
0000-00-00_00-00-00 - Check available WI-FI network: 
0000-00-00_00-00-00 - WiFi networks scan done
0000-00-00_00-00-00 - 6 networks found
0000-00-00_00-00-00 - Nr | SSID                             | RSSI | CH | BSSID             | Encryption
0000-00-00_00-00-00 -  1 | XXXXXXXXX                 |  -68 |  1 | XXX | WPA2
0000-00-00_00-00-00 -  2 | XXXXXXXXX                 |  -69 |  1 | XXX | WPA2
0000-00-00_00-00-00 - SSID:  not found
0000-00-00_00-00-00 - Starting mDNS record: http://prusa-esp32cam.local
0000-00-00_00-00-00 - Starting mDNS OK
0000-00-00_00-00-00 - Init camera lib
0000-00-00_00-00-00 - Init GPIO
0000-00-00_00-00-00 - Init camera module
E (7582) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
0000-00-00_00-00-00 - Camera init failed. Error: 105
0000-00-00_00-00-00 - Reset ESP32-cam!
0000-00-00_00-00-00 - WiFi STA stop
0000-00-00_00-00-00 - WiFi AP stop
ets Jun  8 2016 00:22:57
```

johnyHV, 2024-05-09:
Hello @Airforc . 

You're having an issue with the camera module. In 90% of cases, if there's an issue with initializing the camera, restarting the MCU helps. However, in your case, restarting the MCU doesn't seem to help. Try disconnecting the camera and reconnecting it. Check the correct insertion of the wires from the camera into the connector on the main board.

```
E (7582) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
0000-00-00_00-00-00 - Camera init failed. Error: 105
0000-00-00_00-00-00 - Reset ESP32-cam!
```

bin101, 2024-05-15:
Hi @johnyHV,

I get a similar error:
```
0000-00-00_00-00-00 - Init camera lib
0000-00-00_00-00-00 - Init GPIO
0000-00-00_00-00-00 - Init camera module
E (9312) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (9312) cam_hal: cam_config(385): cam_dma_config failed
E (9312) camera: Camera config failed with error 0xffffffff
0000-00-00_00-00-00 - Camera init failed. Error: ffffffff
0000-00-00_00-00-00 - Reset ESP32-cam!
0000-00-00_00-00-00 - WiFi STA stop
0000-00-00_00-00-ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
E (342) psram: PSRAM ID read error: 0xffffffff
```

After reading this : https://github.com/espressif/arduino-esp32/issues/6209#issuecomment-1168979160
I think my psram is faulty. I got the module fresh in my mailbox today. Is there anything I can do or just RMA the module?

This is the module I ordered: https://www.amazon.de/dp/B0CXXYGNL4

mozgy, 2024-05-15:
try few more things -
- reseat camera connector
- use beefier USB cable and change USB port, some USB ports can't provide enough current and camera init can reset board
- try File > Examples > ESP32 > Camera > CameraWebServer example

bin101, 2024-05-15:
@mozgy 
I can get the camera to work with the example:
```#define CAMERA_MODEL_AI_THINKER```

But even in the example project I get the PSRAM read error:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
E (539) esp_core_dump_flash:��ɕ�dump partition found!
E (539) esp_core_dump_flash: No core dump partition found!
E (362) psram: PSRAM ID read error: 0xffffffff

.
WiFi connected
Camera Ready! Use 'http://192.168.10.14' to connect
```

bin101, 2024-05-15:
@mozgy @johnyHV Found my issue.

My Programmer had a solder bridge which bridged 3.3V and IO16 (PSRAM) on the MCU... So PSRAM wasn't detected. But this means the PSRAM check seems to be faulty here.

johnyHV, 2024-05-16:
@bin101 this is strange issue. Do you have a photo ?

bin101, 2024-05-17:
@johnyHV from the solder bridge? This was just a hardware issue from a bad manufactured programmer board.

cwylie11, 2024-06-03:
Had this same issue! Mine was on the MCU side. The solder was so thick between 3.3V and I16 that I thought it was actually intentional. 

Thanks for the fix, it all works great now!

warwickchapman, 2024-07-25:
I get this loop:

```rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header:```

If I load ESPhome, it works fine.  But the ESP32-CAM version of this project gives me that loop.

Any suggestions?

warwickchapman, 2024-07-25:
I compiled from source 1.1.0rc2 with Arduino IDE and it works.  Not sure why the binaries don't work - maybe something different about this hardware:
[image]
[image]

johnyHV, 2024-07-31:
Hello @warwickchapman . This is strange. HW version looks good. Can you try use for example FW 1.1.0rc1 ? or 1.0.3 ?

warwickchapman, 2024-10-26:
I had a chance to try again and updated to 1.1.0 and:

2024-10-26_20-15-41 - Free RAM: 39988 B, Min: 33968
2024-10-26_20-15-41 - Free PSRAM: 4127268 B, Min: 4111568
2024-10-26_20-15-41 - MCU Temperature: 60.00 *C
2024-10-26_20-15-46 - CardStatus: 1 FileStatus: 1
2024-10-26_20-15-51 - WiFi status: Wifi connected. SSID: Sensible IOT, BSSID: 7A:9A:18:07:06:1C, RSSI: -93 dBm, IP: 192.168.0.235, TX power: 19.5dBm
E (314665) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (314665) task_wdt:  - CaptureAndSendP (CPU 0)
E (314665) task_wdt:  - SystemNtpOtaUpd (CPU 0)
E (314665) task_wdt:  - IDLE0 (CPU 0)
E (314665) task_wdt: Tasks currently running:
E (314665) task_wdt: CPU 0: CaptureAndSendP
E (314665) task_wdt: CPU 1: IDLE1
E (314665) task_wdt: Aborting.
E (314665) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4000bfed:0x3ffe90d0 0x40091ec2:0x3ffe90e0 0x40090a86:0x3ffe9100 0x4012b656:0x3ffe9140 0x4012b98e:0x3ffe9160 0x4012c03d:0x3ffe9180 0x4012c1a4:0x3ffe91b0 0x4012c22e:0x3ffe9200 0x4011797f:0x3ffe9220 0x40118ee9:0x3ffe9260 0x40118f96:0x3ffe92c0 0x40115d49:0x3ffe92e0 0x4000bdbb:0x3ffe9300 0x400017ea:0x3ffe9320 0x401662b1:0x3ffe9340 0x4016020d:0x3ffe9360 0x40160ac1:0x3ffe9380 0x4016127a:0x3ffe93f0 0x400fa9cf:0x3ffe9410 0x400f9dbd:0x3ffe9430 0x400f9d0f:0x3ffe9450 0x401bcc1d:0x3ffe9480 0x400e22ca:0x3ffe94b0 0x400e29bd:0x3ffe9660 0x400e2a76:0x3ffe96e0 0x400e6645:0x3ffe9720 0x40093062:0x3ffe9780

ELF file SHA256: 6fadd5aab0aba042

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15488
load:0x40080400,len:4
load:0x40080404,len:3180
entry 0x400805b8
----------------------------------------------------------------
Start MCU!
Prusa ESP32-cam https://prusa3d.cz
SW Version: 1.1.0
Build: Aug  5 2024 18:24:22
----------------------------------------------------------------