prusa3d/Prusa-Firmware-ESP32-Cam #14: README correction

mozgy, 2024-05-06:
Both board version can be programmed with CH640 add-on MB, it's just that with 'older' 4 SMDs CAM version you need to press *both* switches on the add-on MB and the reset one on the CAM simultaneously to enter flash mode. On the 6 SMDs CAM version all that works automatically through USB connection. To make things a bit more interesting, there is a new add-on MB with dupont extension that works fine with 'older' 4 SMDs CAM version *BUT* does not work with 'newer' 6 SMDs CAM. To summarize it's not CH640 fault but programmer board designers and RTS/DTR connections ..

johnyHV, 2024-05-09:
Yes it's true. to simplify the situation, it was written like this. I can try to reformulate it, but it must be comprehensible even for less proficient users

mozgy, 2024-05-10:
actually it even gets worse .. I've just received ESP32S3-CAM and it's vice-versa success on MBs .. I'll write some markup text over the weekend ..

closed 2024-05-13 by johnyHV