prusa3d/Prusa-Firmware-ESP32-Cam #16: cfg.h: no such file or directory

jakefreese, 2024-05-10:
I have been attempting to get my cameras up and going and can not get them to compile.  I have tried with the Arduino compiler and VSC with Platform IO and get the same errors.  I have the ESP32 by Espressif Systems 2.0.15 installed along with the 4 required libraries.  It is probably an error on my end that I have not been able to identify.

Thank you!

johnyHV, 2024-05-10:
Sorry, but I don't have problem with compilation in the Arduino IDE 2.3.2, and in the Visual studio code with Arduino plugin. Can you send me output from the console  ?

jakefreese, 2024-05-11:
From VSC with Platform IO plugin, since I copied the error in to this window I did notice the two warnings that it can not find two packages.  This could be my issue.  I will work more on it tomorrow, thank you!

Library Manager: Installing syncTCP
Warning! Could not find the package with 'syncTCP' requirements for your system 'windows_amd64'
Library Manager: Installing ESPAsyncWebSrv
Warning! Could not find the package with 'ESPAsyncWebSrv' requirements for 
your system 'windows_amd64'
Found 35 compatible libraries

I do have the "Arduino ESP32 Boards by Arduino 2.0.13 " and "ESP32 by Espressif 2.0.16" installed
   
`Executing task in folder PRUSA_ESP32_CAMERA: C:\Users\jaker\.platformio\penv\Scripts\platformio.exe run 

Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
---------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.4.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing syncTCP
Warning! Could not find the package with 'syncTCP' requirements for your system 'windows_amd64'
Library Manager: Installing ESPAsyncWebSrv
Warning! Could not find the package with 'ESPAsyncWebSrv' requirements for 
your system 'windows_amd64'
Found 35 compatible libraries
Scanning dependencies...     
Dependency Graph
|-- ArduinoJson @ 7.0.4    
|-- ArduinoUniqueID @ 1.3.0
|-- ESPmDNS @ 2.0.0        
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32cam\src\main.cpp.o
Building .pio\build\esp32cam\bootloader.bin
Generating partitions .pio\build\esp32cam\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32cam\lib5b9\ArduinoUniqueID\ArduinoUniqueID.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32cam\libf94\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32cam\lib1d2\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\IPAddress.cpp.o
src/main.cpp:53:10: fatal error: cfg.h: No such file or directory

*************************************************************
* Looking for cfg.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:cfg.h"
* Web  > https://registry.platformio.org/search?q=header:cfg.h
*
[... 195 lines cut ...]
Compiling .pio\build\esp32cam\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-bt.c.o
src/main.cpp:53:10: fatal error: cfg.h: No such file or directory

*************************************************************
* Looking for cfg.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:cfg.h"
* Web  > https://registry.platformio.org/search?q=header:cfg.h
*
*************************************************************

 #include "cfg.h"
          ^~~~~~~`

From Arduino IDE 2.3.2

`C:\Users\jaker\OneDrive\Documents\Arduino\Prusa_ESP32_cam\Prusa_ESP32_cam.ino:53:10: fatal error: cfg.h: No such file or directory
   53 | #include "cfg.h"
      |          ^~~~~~~
compilation terminated.
exit status 1

Compilation error: cfg.h: No such file or directory`

johnyHV, 2024-05-13:
Hello. Did you try to install the missing packages? It helped ?

jakefreese, 2024-05-20:
I still have not been able to figure out where the packages need to be in PlatformIO and have them linked.  I am going to keep looking today.  I have the file with all the packages in the "include" folder, I think I need to add something to the platformio.ini file to point the where the files are located.

closed 2024-05-20 by jakefreese