prusa3d/Prusa-Firmware-ESP32-Cam #17: Live view and Snapshot URL

cyberjak2k, 2024-05-10:
Just more of a question 
I am not running prusa connect on my i3 MK3S+ but I am using octoprint on a Pi Zero 2w plugged into the Einsy board. I was wondering , 
Does this offer a live view stream url, as i think i've figured out a way for it to poll the snapshot image and it causes a refresh of the image by the request but just figured i'd ask

johnyHV, 2024-05-10:
You can obtain the image in two ways.

The first method is to manually request the capture by calling http://ip/action_capture and then downloading it from http://ip/saved-photo.jpg.

The second option is to open the current stream from http://ip/stream.mjpg. Unfortunately, the stream is currently unreliable, and the camera may freeze during its use. I have already created a fix for this issue. I need to test the fix more thoroughly before publishing it.

cyberjak2k, 2024-05-10:
Awesome as eventually I'll try put the Prusa connect just with different SD
card hahah
I was using a multi steam one previously that seems to work well. I wish I
knew how to code better

johnyHV, 2024-05-13:
@cyberjak2k I recently added a new release 1.0.2-rc1. In this version, the problem with the stream is fixed. You can test it at http://IP/stream.mjpg

cyberjak2k, 2024-05-13:
Thanks, I'll check it out!  Awesome!

cyberjak2k, 2024-05-13:
Ok, just used the OTA update instead of downloading and recompiling. And it's holding ROCK Solid!
Hoping that at some point the ability to just point it at a URL and every time that you URL is refreshed it takes a fresh snapshot instead of having to do a manual refresher snapshot and then download said snapshot. But this is a massive jump and it works great Thank you

closed 2024-05-13 by cyberjak2k

BugsBunny1403, 2024-05-14:
Hello, can you add the link for stream in config page of esp32 cam ?