prusa3d/Prusa-Firmware-ESP32-Cam #19: Prusa Connect Camera name resets after Camera reboot

jimwsmith, 2024-05-11:
Every time Camera reboots, the displayed name in PrusaConnect resets to ESP32-CAM.
I manually edit it and it's fine until the camera reboots

jimwsmith, 2024-05-13:
Would be great to have ability to set Camera name in ESP32-cam.

johnyHV, 2024-05-16:
@jimwsmith The current FW 1.0.2-rc2 uses the mDNS record as the camera name in PrusaConnect. I think it is the correct use of one variable for 2 functions.

jimwsmith, 2024-05-17:
> @jimwsmith The current FW 1.0.2-rc2 uses the mDNS record as the camera name in PrusaConnect. I think it is the correct use of one variable for 2 functions.

Works great (and saves creating a new setting too)

johnyHV, 2024-05-17:
Great. Thank you for the improvement suggestion :) .

closed 2024-05-17 by johnyHV