prusa3d/Prusa-Firmware-ESP32-Cam #20: rotation of the image

Panik68, 2024-05-11:
Could the rotation of the image in 4 steps 0(360), 90, 180, 270° be included in the next update?

mozgy, 2024-05-11:
you cannot rotate 90° or 270°, only vertical and horizontal flips so 180° would be both flips active ..

johnyHV, 2024-05-13:
The camera module does not support rotating the photo by 90° and 270° degrees. I did not find a solution to rotate the image by 90° and 270° .

Panik68, 2024-05-15:
Thank you very much for your feedback. I have also checked again and it is indeed not possible to rotate the image. Apparently this is due to the camera module itself. What a pity.

nvtkaszpir, 2024-05-16:
it would be possible to add jpeg exif metadata and there you can define image rotation, so this way esp32 does not have to do any actual rotations, the client would be aware the image needs to be rotated, and modern browsers already do that

johnyHV, 2024-05-16:
@nvtkaszpir Hmm this is good idea. I've never tried anything like that and ESP32-cam have a low RAM memory. but I will try to see if it is possible

johnyHV, 2024-05-19:
@nvtkaszpir A perfect idea with EXIF. The OV2640 does not generate EXIF data in the JPG file. However, I managed to add EXIF to the photo, and rotation works. I am still testing it, but if the tests go well, this change will appear in firmware version 1.0.3.

johnyHV, 2024-05-27:
@nvtkaszpir new FW v1.0.3-rc1 has added a support for EXIF and image rotation :)

nvtkaszpir, 2024-05-28:
ha, awesome!

closed 2024-05-30 by johnyHV