prusa3d/Prusa-Firmware-ESP32-Cam #22: Add support for ESP32-WROVER board

firstmustburn, 2024-05-11:
I create pull request #21.

This adds support for the Freenove ESP32-WROVER board. Everything in the original firmware works fine except that the GPIO must be remapped. So there are #if statements to make this change in two files. I also wrote a detailed readme for building and installing the code on the board.

The main goal here is to have the ESP32-WROVER benefit from bug fixes and improvements. Putting #if switch statements around the GPIO definitions doesn't seem that elegant or scalable, but I could not find a better strategy anywhere. Also, it seems like the Arduino IDE doesn't have a way to add compiler flags, so the #define for the flag has to be set in each file. If there's a different, preferred way of doing this, I'm happy to rework the code and update the PR.

johnyHV, 2024-05-13:
Great. I need to order an ESP32-Wroover, test its stability, and then optimize the code to be more user-friendly and usable.

johnyHV, 2024-06-29:
Hello @firstmustburn support for the ESP32-WROVER is already added. Can you tested it please ?

firstmustburn, 2024-06-30:
It will probably be a week or so before I will have access to the hardware to test, but I will check it out and post back.

johnyHV, 2024-06-30:
@firstmustburn Sure. Thank you. I tested FW on the board ESP32-WROVER, and everything works for me.

firstmustburn, 2024-07-17:
Hello @johnyHV, I tested the current master branch and it worked fine.  I made some detailed below.  I think the important parts are:
- add some documentation on how to get the libraries that are not available in the arduino library manager (or, like the case of AsyncTCP, it is the wrong repo that comes up in the search.  The git incantations I have in the notes might be helpful as a script.
- Add a big warning symbol or something to the part that says you have to modify the source code to build in the arduino IDE.  this was easy to miss, and if someone doesn't know how to troubleshoot it, they probably won't be able to figure out what's wrong.
- For troubleshooting, note that after you reprogram, you may have to recreate the camera in PrusaConnect if it is getting 400 errors.

---

`git clone git@github.com:prusa3d/Prusa-Firmware-ESP32-Cam.git`
Open arduino IDE 2.3.2
Open `ESP32_PrusaConnectCam/ESP32_PrusaConnectCam.ino` in arduino IDE

Install libraris:
Tried to update libraries as describe here: `https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/blob/dc4a0a6d40dc757f3900cafc76c354c76ad864cf/README.md?plain=1#L5`
but many of them are not available in the arduino IDE:

- set my sketchbook folder in Ardunio IDE to `path/to/Prusa-Firmware-ESP32-Cam/ESP32_PrusaConnectCam`

```sh
cd path/to/Prusa-Firmware-ESP32-Cam/ESP32_PrusaConnectCam
mkdir libraries
git clone git@github.com:mathieucarbou/AsyncTCP.git --branch v3.1.4 --single-branch
git clone git@github.com:mathieucarbou/ESPAsyncWebServer.git --branch v3.0.6 --single-branch
git clone git@github.com:bblanchon/ArduinoJson.git --branch v7.1.0 --single-branch
git clone git@github.com:ricaun/ArduinoUniqueID.git --branch 1.3.0 --single-branch
git clone git@github.com:RobTillaart/DHTNew.git --branch 0.4.20 --single-branch
```

Set the Arduino options as described at: `https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/blob/master/doc/ESP32-Wrover-dev/README.md#how-to-compile-software-in-the-arduino-ide`

-  Note:  there was no option for `Tools -> CPU Frequency -> 240MHz (WiFi/BT)`

Run `Verify/compile` in Arduino IDE

Got this error once, re-ran it and it compiled without errors

```
Sketch uses 1591957 bytes (121%) of program storage space. Maximum is 1310720 bytes.
Global variables use 56248 bytes (17%) of dynamic memory, leaving 271432 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board
```

After programming, the board was intermittenly resetting.  Connecting the serial monitor in the arduino IDE, the log ended with:

```
E (8229) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
0000-00-00_00-00-00 - Camera init failed. Error: 105
0000-00-00_00-00-00 - Reset ESP32-cam!
E (16866) wifi:NAN WiFi stop
ets Jul 29 2019 12:21:46
```

Found this instruction in the readme, consider making it bigger!
```
It is necessary to enable support for the correct board version in the file mcu_cfg.h after line 16.
```

Modified the mcu_cfg.h to set:

```c
#define AI_THINKER_ESP32_CAM           false
#define ESP32_WROVER_DEV               true
#define CAMERA_MODEL_ESP32_S3_DEV_CAM  false
#define CAMERA_MODEL_ESP32_S3_EYE_2_2  false
#define CAMERA_MODEL_XIAO_ESP32_S3_CAM false
#define CAMERA_MODEL_ESP32_S3_CAM      false
```

Rebuilt and reflashed, everything booted fine.

Connected to the AP and configured the WIFI.

Put the old device token in, and was getting 400 and 403 errors.

Deleted the camera in prusa-connect, created a new camera, and everything worked fine.

firstmustburn, 2024-07-17:
@johnyHV One more thing:  is there any plan to set up CI to build the firmware binaries so they can just be downloaded?  If you want help containerizing the build, I could work on that.

johnyHV, 2024-07-18:
@firstmustburn I understand what you are focusing on and what your goal is. However, the primary purpose is not for the user to compile the source code themselves. The user should download the precompiled binary from the release. Compiling their own source code is only intended for experienced users. I don't want to unnecessarily complicate the guide with instructions for installing libraries and compiling. It could make the README confusing. Nevertheless, I will incorporate some of your ideas into the README and try to improve it. Each board has its specific settings in the Arduino IDE. Therefore, a separate guide is created for each board.

Thank you, I plan to create a pipeline (i have skill only with bitbucket pipeline, but it's ist very similar). That's why I have created the compilation scripts. Unfortunately, I haven't gotten around to it yet. However, I think for the pipelines on GitHub its neccesary pay extra money.

I'm happy the new firmware is working.

closed 2024-07-30 by johnyHV