prusa3d/Prusa-Firmware-ESP32-Cam #23: Does this work for ESP-EYE too ?

piwi1263, 2024-05-12:
Hi I've got an ESP-EYE V2.1 Devkit and as far as I know the camera is an OV2640 but any xp on this devkit ?

firstmustburn, 2024-05-12:
I was able to get the code working with an ESP32-WROVER board (see my pull request #21 for the files I changed).  

Based on the [schematic for the ESP_EYE board](https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-EYE-MB_20210913_V2.1.pdf), it looks like the status LED is GPIO3.  This [page ](https://github.com/espressif/esp-who/blob/master/docs/en/Camera_connections.md) that documents the camera pin connections, so you could use that to modify the files along the same lines I did.

You won't have SD card support or be able to add a flash, since there is no free GPIO, but it should be easy enough to try.  You'll see in the serial monitor if the board is working.  If it's resetting, it probably means the pins are not all correct.

johnyHV, 2024-05-13:
@firstmustburn  Attention, it's not that simple. ESP32-WROOVER is built on the ESP32 processor, and ESP32-CAM is also built on the ESP32 processor. However, there are boards built on the ESP32-S3. With the ESP32-S3 processor, I didn't have good results. Maybe it was just my dev board that was faulty. But ESP-EYE V2.1 seems to use the ESP32 processor. 

@Piwi1263 Compatibility needs to be tested. Currently we don't have SW support for ESP-EYE 2.1

mozgy, 2024-05-13:
@johnyHV ESP32S3-CAM works fine for me with #define CAMERA_MODEL_ESP32S3_EYE (minus flash led)

johnyHV, 2024-05-13:
@mozgy Can you send me the version of the board you are using?

mozgy, 2024-05-13:
Here it goes - 
[image]
haven't got time to figure out flash LED yet and yes I have resoldered jumper for antennae

johnyHV, 2024-05-14:
@mozgy  Oh really ? Great! Where did you buy this board ? Can you send me camera pin mapping from file from the file Camera_cfg.h ?

mozgy, 2024-05-14:
@johnyHV as I wrote before, it's copy from CAMERA_MODEL_ESP32S3_EYE. Only thing different is ```define LED_GPIO_NUM    48``` with caveat that it is actually RGB LED working with Adafruit_NeoPixel library .. and ofc it's from Aliexpress ..

johnyHV, 2024-06-29:
Hello @piwi1263 support for the ESP32-S3-EYE 2.2 is already added. Can you tested it please ?

closed 2024-07-30 by johnyHV