prusa3d/Prusa-Firmware-ESP32-Cam #24: Is it possible to add a solution to activate the LED from prusa connect ?

BugsBunny1403, 2024-05-14:
Good morning.
I noticed that Prusa Connect retrieved information from the ESP 32 Cam (IP, MAC Addr, SSID, camera name, etc.) Is it possible to add a solution to activate the LED continuously or by flash shooting from prusa connect. Maybe see more options to be able to control the cams remotely. I suspect that this involves changes on both sides. But not having any ideas to share my request for the prusa connect, I would like to post here and if you have the contact or the information on where to contact them. In advance, thank you for the work already accomplished.

Zero-Circle, 2024-05-18:
@BugsBunny1403, this is a fantastic suggestion. 
@johnyHV i think it should have a switch in connect for turning it on and off and also a setting to turn on everytime an image is captured. OFC i understand this is not part of the scope of esp32 cam but these are now closely working together with prusa which is fantastic for the community.

johnyHV, 2024-05-19:
@Zero-Circle Yes I agree. It would be nice to control the camera from connect. I believe that we will find a solution to control the camera from prusa connect in the future. I will open this topic at the next meeting with the Prusa team

BugsBunny1403, 2024-05-19:
I am delighted that my proposal is followed and acclaimed like this. I think that esp cam management with prusa connect would be useful because we don't necessarily have access to the cam when we are far away. Therefore, you would need to have the main settings (continuous LED, and shooting, exposure time, etc.).
PS: maybe I should open another ticket. I just had a problem, the ESP couldn't connect to the wifi, it started to heat up and the AP suddenly started to glitch. I think it could be interesting to have wifi management for two ssids depending on the connection quality.

johnyHV, 2024-05-19:
@BugsBunny1403 please can you make new ticket with two SSID ? It's very good idea. Thank you

johnyHV, 2024-05-31:
@BugsBunny1403 currently I don't have any new.  We are discussing with Prusa team about this possibility

VibroAxe, 2024-06-15:
@johnyHV this is a great suggestion. Mainly because the prusa connect can then handle options for turning the led on ONLY when printing. Currently my led is now flashing in the office 24hrs which is upsetting the family! 

(Still a great project btw, and much more reliable than when it started out. Well done on the prusa transition)

johnyHV, 2024-08-22:
Thank you @VibroAxe . We are still discussing with Prusa team about this possibility

Madad1996, 2024-10-06:
any News on this?
would be great to only use the light while printing

noisufnoc, 2024-10-21:
fwiw, you can toggle the LED from the following api endpoint `http://camera.ip.address/action_led?/led=true` so if you are handy you might be able to script something to check prusa connect if your printer is printing and then toggle the LED that way.