prusa3d/Prusa-Firmware-ESP32-Cam #27: Camera Snapshots Based on Printer Activity

der-eine, 2024-05-18:
Hello and thank you for this great project.

I noticed that the camera always takes snapshots as soon as it is turned on. Is it possible to link this to the state of the printer so that pictures are only taken when the printer is actually working? Or have I overlooked a relevant setting?

Best regards,
der-eine

der-eine, 2024-05-19:
@johnyHV 
All right. Thanks for the info. I was hoping, since you are working very closely with Prusa, that this would be an option to query the status via the API. 

Another question regarding snapshots: will there ever be a direct snapshot URL? To be able to send a snap via Telegram, for example? 
I already know the option of „calling http://ip/action_capture and then downloading it from http://ip/saved-photo.jpg.“

johnyHV, 2024-05-20:
@der-eine I will open this topic at the next meeting with people from Prusa. It would be best to have an API from Prusa Connect. I believe that the API for the Prusa Connect about printer status will be added in the near future. It would also be good for them, so that they don't have such a big photo traffic.

You mean sending photo from the camera to telegram ? Prusa Connect have added this feature for printer status. But it is just text message without photo.

der-eine, 2024-05-21:
@johnyHV That's a great idea. I also think that an API would be helpful for many things.

Thanks for the information. Correct, unfortunately, it currently only supports starting, stopping, and error messages. An option to send a photo via Telegram every x percent or at layer X would certainly be useful.

Anheledir, 2024-05-27:
If Prusa Connect would support webhooks with a custom endpoint, it could be possible to start / stop making and sending regular camera updates based on these. But that would make it necessary to make the camera module accessible from the internet - which i don't really like from a security point of view.

The other way around the camera would need to ask the connect api in regular intervals for a printers status. That would reduce the traffic as that api call has a way smaller size than an image, but still has a significant impact for the server's resources. 

Not a big fan of either variant. Some other ideas to think about:
- establishing a WebRPC connection for communication (additional implementation in prusa correct necessary)
- controlling the board via serial connection, connecting it to the printer directly, e.g. via the already existing ESP-WiFi board as UART adapter (would need hw/fw modifications on the printer, though)
- Adding a hw switch or sensor to the camera board that could be triggered by the extruder or print-bed moving to a certain position via gcode (more hw modifications and adding of additional gcode to all prints in the slicer, though)
- Analyzing of the camera picture locally on the board to recognize when the print-head starts moving, maybe with a one-time calibration/setup in the web-ui (no additional api or hw necessary, but could be unreliable)

Just a few thoughts about this ... maybe there are better ideas 💡🤔🤪

VibroAxe, 2024-06-15:
Given the camera is likely on the same network as the printer, it could query the prusalink API running locally on the printer for this instead of going via prusa connect

mikromcz, 2024-06-16:
> Given the camera is likely on the same network as the printer, it could query the prusalink API running locally on the printer for this instead of going via prusa connect.
> 
> The best option I can think of here would be to have different refresh rates based on printer state. So if the printer isn't moving it sends an image once every X minutes, once printing starts it sends every y seconds

Connecting to the printer API sounds like a good idea to me. I already did my research and here you can see what is available on the API. (I'm using it for Homepage dashboard) https://github.com/mikromcz/homepage/blob/main/widget_prusalink.yaml
[image]

donmaro, 2024-07-17:
I've connected my printer to prusa connect and prusa connect to home assistant. In home assistant i get state changes and i can call API's. Is there an API on the ESP i can call to start and stop the filming / photos?

JWindy, 2024-08-26:
Hey, 
I setup my camera during the last few days and I'm very happy with the result! Thanks for the effort!

Although a trigger through the API would be most convinient, the trigger could also be provided by connection a gpio pin of the microcontroller to a gpio pin of the new hackboard. This might be a workaround until (hopefully) the api is extended by this feature. What do you think?

Windy

ZevEisenberg, 2024-10-22:
Just came here to +1 this! I have no SD card in the camera, and I'm not planning to use it for time lapses; just for checking in on the printer. Therefore, I don't need it taking pictures and heating up the components unless I'm actively looking at the feed.

tonysurma, 2025-06-25:
> I've connected my printer to prusa connect and prusa connect to home assistant. In home assistant i get state changes and i can call API's. Is there an API on the ESP i can call to start and stop the filming / photos?

This would be ideal, or even the ability to set the interval to 0 or something to tell it to stop sending.  I use home assistant to do similar to what you are saying for unifi protect cameras over my printers so they only record when not idle.

Also I modded a [script for raspberry pi in the XL Cam printables model](https://github.com/tonysurma/3dprinter-camera-streamer) to check prusalink before it uploads.  Was just thinking I could stop having the ESP32 cam send the photo directly and instead mod my script on a raspberry pi on the network to check the printer via prusalink then grab the snapshot from the ESP32 cam and upload.