prusa3d/Prusa-Firmware-ESP32-Cam #28: Brightness

ice987987, 2024-05-20:
Great work! First I want to thank you for this code!

I just installed 1.0.2-rc2 (Software build | May 20 2024 13:35:00) and played a bit around with the settings. I saw, that "Brightness" has no influence on the picture, could you please check and confirm?

Many thanks

johnyHV, 2024-05-20:
Thank you. I will check the functionality of the brightness setting

ice987987, 2024-05-24:
interestingly, the displayed photo on the prusa esp32-cam site is not the same as the picture shown on https://connect.prusa3d.com/printer/.../cameras

johnyHV, 2024-05-31:
photo on the camera WEB page is not automatically reload. It would be an unnecessary load on the camera. Therefore, it may seem that the photographs do not match.

johnyHV, 2024-06-01:
@ice987987 I see same issue with configuration brightness, contrast, saturation. But source code looks good. Correct parameters and correct values ​​are set. But as if the configuration is not applied in the camera. So currenty I don't know what is bad. 

But currently you can tested disable "automatic gain control" and change the value "Automatic gain control level" for change the brightness

[image]

ice987987, 2024-06-02:
Thank you for checking the code. I tried various combinations of the settings and it looks like you mentioned above. `Automatic gain control` should be disabled, and then it looks good.

closed 2024-06-02 by ice987987