prusa3d/Prusa-Firmware-ESP32-Cam #29: ESP32 S3 CAM support

mozgy, 2024-05-22:
Hello, this setup works for me ->
```
   @file Camera_cfg.h
// OV2640 camera module pins (CAMERA_MODEL_ESP32S3_CAM)
#define PWDN_GPIO_NUM -1
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 15
#define SIOD_GPIO_NUM 4
#define SIOC_GPIO_NUM 5

#define Y2_GPIO_NUM 11
#define Y3_GPIO_NUM 9
#define Y4_GPIO_NUM 8
#define Y5_GPIO_NUM 10
#define Y6_GPIO_NUM 12
#define Y7_GPIO_NUM 18
#define Y8_GPIO_NUM 17
#define Y9_GPIO_NUM 16

#define VSYNC_GPIO_NUM 6
#define HREF_GPIO_NUM 7
#define PCLK_GPIO_NUM 13

#define LED_GPIO_NUM 48 // RGB LED NeoPixel
```

and this is the code that drives the LED with no additional libraries, just <Arduino.h> ->

```
#define RGB_BUILTIN LED_GPIO_NUM

void loop() {
#ifdef RGB_BUILTIN

// This does NOT WORK
  digitalWrite(RGB_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  digitalWrite(RGB_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);
// This does NOT WORK

// This WORKS
  neopixelWrite(RGB_BUILTIN, RGB_BRIGHTNESS,RGB_BRIGHTNESS,RGB_BRIGHTNESS);   // Turn the RGB LED white
  delay(1000);
  neopixelWrite(RGB_BUILTIN, 0,0,0);    // Turn the RGB LED off
  delay(1000);

  neopixelWrite(RGB_BUILTIN,RGB_BRIGHTNESS,0,0); // Red
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,RGB_BRIGHTNESS,0); // Green
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,RGB_BRIGHTNESS); // Blue
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,0); // Off / black
  delay(1000);
// This WORKS

#endif
}

```

unfortunately full white output is not that bright as in original CAM,
have fun

__
Mozz

johnyHV, 2024-05-23:
Hi. Thank you very much. Could you please send me the board version? Or a photo? There are several ESP32-S3-cam boards

mozgy, 2024-05-23:
Hey hey,
I'm trying to do the pull req but since you're using FLASH_LED pin as a HW boolean flag all around the code, it's a bit challenging. IMNHO new variable might be needed for that. Here's the picture - 
[image]
pinout is the same so it works on 1st version of MB and it has USB-C

johnyHV, 2024-05-27:
Hi @mozgy . I have ordered this board. if it works, I'll edit the source code and add support for it.

mozgy, 2024-05-27:
Awesome,
@johnyHV take a look at my esp32-cam repo, I have all this working ->
```
// Select camera model
// #define CAMERA_MODEL_AI_THINKER
// #define CAMERA_MODEL_XIAO_ESP32S3
#define CAMERA_MODEL_ESP32S3_CAM
```
it might help ..

johnyHV, 2024-05-27:
@mozgy Greate! XIAO ESP32-S3 works ? Okay, this is new information for me. 

Currently I have on the table
- ESP32-CAM
- ESP32-WROVER-DEV
- ESP32-S3-box

and it will come to me in the near future
- ESP32-S3-CAM
- ESP-EYE V2.1 
- 
 and something else.
 
 But some cameras don't have an SD card, others don't have an LED, and so on. So I have to come up with the universality of the code

ulab, 2024-05-27:
There is no way to detect the different versions?

I wonder if you could have a select box for the model in the settings and start with some common code, but enable specifics if people change the selector?

To be fair I already have issues figuring out what board I have when I just get some random ESP32-Cams from China.

Anheledir, 2024-05-27:
> There is no way to detect the different versions?

I like that idea somewhat, as it would have it easier for regular users. 🤔

To detect the specific ESP32 board the `esp_chip_info_t` structure and `esp_chip_info` function provided by the ESP-IDF could be used. This method would give us detailed information about the chip, such as the number of cores, features like WiFi, Bluetooth, and BLE, silicon revision, and flash size.

Here’s a sample code snippet for detecting the ESP32 chip I used in a different project a while ago:

```cpp
#include <Arduino.h>
#include "esp_system.h"

void setup() {
  Serial.begin(115200);
  
  esp_chip_info_t chip_info;
  esp_chip_info(&chip_info);

  Serial.printf("This is an ESP32 chip with %d CPU cores, WiFi%s%s\n",
                chip_info.cores,
                (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
                (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

  Serial.printf("Silicon revision %d\n", chip_info.revision);

  Serial.printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
                (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

  // Additional detection logic based on chip features, GPIO pins, etc.
}
```

Some other features like the presence of an LED flash or GPIO config would be some additional logic to detect specific boards like ESP32-CAM, ESP32-S3-CAM, and others by checking specific GPIO configurations and unique features of these boards. I only have the regular ESP32-Cam boards myself, so someone with the other boards would need to help how to identify them reliably. 

```cpp
void detectBoard() {
  esp_chip_info_t chip_info;
  esp_chip_info(&chip_info);

  // Example logic to detect ESP32-CAM
  if (chip_info.cores == 2 && chip_info.revision == 1) {
    // Additional checks for ESP32-CAM specific pins and features
    if (digitalRead(GPIO_NUM_4) == HIGH && digitalRead(GPIO_NUM_5) == LOW) {
      Serial.println("Detected ESP32-CAM");
    }
  }

  // Example logic to detect ESP32-S3-CAM
  if (chip_info.model == CHIP_ESP32S3) {
    // Additional checks for ESP32-S3-CAM specific pins and features
    Serial.println("Detected ESP32-S3-CAM");
  }

  // Add more detection logic for other boards as needed
}
```

Using the combination of `esp_chip_info` and board-specific checks, it would be possible to identify the board the fw is running on. Would need some additional refactoring for the defining / setting of variables, too. Just leaving this comment here as draft on further developing.

johnyHV, 2024-05-31:
@Anheledir this looks wery good. But it will probably be easier to solve this with a precompiler and different FW for severals cameras. Different camera modules use different versions of ESP32/ESP32-S3 modules, and the firmware between them was not fully compatible. I'm currently analyzing it and trying to find the first functional solution. However, I need to complete all the more critical tasks first, and then I'll tackle this problem :) ( I work on it in just my free time ). I believe that next week I will receive more camera modules so that I can test the firmware on multiple cameras.

Anheledir, 2024-05-31:
@johnyHV That's okay, just as you say there are more critical things to do first and it's just a free-time project. I'm planning to help out now and then with some contributions, depending on my own free time. So just let me know when you need help with something specific 😊👍🏻

mozgy, 2024-05-31:
ESP32 S3 CAM definitely should have separate FW, currently I'm building it with `-DARDUINO_ESP32S3_DEV` and `default_16MB.csv` partition table.

johnyHV, 2024-06-27:
@mozgy please, where you are found schematic for this ESP32-S3-CAM board ?

mozgy, 2024-06-27:
@johnyHV I didn't, it came with completely the same MB so I presumed it should work. I did the continuity test to be sure. Please *BE* aware that pinout functions are *NOT* on the same IO pins on S vs S3, ie UART0 on S is IO1/IO3 but on S3 it's IO43/IO44.

johnyHV, 2024-06-27:
@mozgy Thank.s I have the same problem. I can't find the pinout and schematic for this board. I asked the seller for the schematic, and I'll see what they respond. I need the pinout for the micro SD card, the STATUS LED on the board, the FLASH LED, and so on. Without the schematic, I'll have to search for it manually... In terms of documentation, this board is very poor.

mozgy, 2024-06-27:
As I wrote at the start, for S3 CAM FLASH LED is IO48, ->
```
#ifdef CAMERA_MODEL_AI_THINKER
  pinMode( FLASH_LED, OUTPUT );
  digitalWrite( FLASH_LED, HIGH );
#endif
#ifdef CAMERA_MODEL_ESP32S3_CAM
  neopixelWrite( FLASH_LED, RGB_BRIGHTNESS, RGB_BRIGHTNESS, RGB_BRIGHTNESS );
#endif

```

 for SDCard init use this ->
```
#ifdef ARDUINO_ESP32S3_DEV
#define SD_MMC_CMD 38
#define SD_MMC_CLK 39
#define SD_MMC_D0  40
    SD_MMC.setPins( SD_MMC_CLK, SD_MMC_CMD, SD_MMC_D0 );
#endif
  if( !SD_MMC.begin( "/sdcard", true ) ) { // slow 1bit mode
...
```

johnyHV, 2024-06-29:
Great. Thanks @mozgy. And do you know the pin for the STATUS LED? my board will arrive by post within 1-2 weeks. Currently is added limited support for the ESP32-S3-CAM, without neopixel LED

mozgy, 2024-06-30:
Sorry @johnyHV haven't bothered with that, all my Cams are inside 3d printed boxes ..

johnyHV, 2024-07-12:
Hello @mozgy in the new FW version 1.1.0-rc2 is added support for ESP32-S3-CAM board.

closed 2024-07-18 by johnyHV