prusa3d/Prusa-Firmware-ESP32-Cam #33: User Friendly Upload

ALD-Models, 2024-05-28:
To make the install process easy for users I've created a website that users can go to to easily install the firmware. It uses esp-web tools and contains a user friendly interface which also show errors and logs

This is the website: https://ald-models.github.io/Prusa-ESP32-Cam-Web-Upload/ 

Hopefully this should be useful!

Jake

BugsBunny1403, 2024-05-28:
It's great, I'm going to test it right now and it's easy to upload the firmware. Just one problem, wifi management by esp web tool is apparently unavailable. I think it has something to do with an improv wifi and web serial function.

ALD-Models, 2024-05-28:
Thanks. I've not yet enabled the WiFi setup but i might play around with it tomorrow!

Jake

johnyHV, 2024-05-29:
Awesome! It is possible in the future added support for multiple different MCU version ? Or configuration the WiFi network, token and something else ?

johnyHV, 2024-05-29:
I see that currently, the binary files are stored in your GIT repository. This can raise suspicion and distrust. Would it be possible to use the binary files directly from the release?

BugsBunny1403, 2024-05-29:
If you create own site with esp tool yes. Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Miroslav Pivovarsky ***@***.***> Date : 29/05/2024  14:16  (GMT+01:00) À : prusa3d/Prusa-Firmware-ESP32-Cam ***@***.***> Cc : BugsBunny1403 ***@***.***>, Comment ***@***.***> Objet : Re: [prusa3d/Prusa-Firmware-ESP32-Cam] User Friendly Upload (Issue #33) 
I see that currently, the binary files are stored in your GIT repository. This can raise suspicion and distrust. Would it be possible to use the binary files directly from the release?

ALD-Models, 2024-05-29:
Yes that is a good point and I guess it would also make sense to keep with the latest update.

I'm not quite sure how to make the link 

Is their a easy method to do this?

Jake

cwylie11, 2024-06-03:
can't connect to the esp's wifi when using this method of installing onto the ESP32. Is the password supposed to be the same?

The ESP network shows up, but just says "can't connect to this network" when it tries to authenticate and connect.

Any suggestions?

ALD-Models, 2024-06-04:
The password should be 12345678. If the problem persists, try turning mobile data off or connecting via a laptop/computer.

Jake

Anheledir, 2024-06-04:
You should either clear the esp32 by default, or add a separate button to do so. The error that you cannot login to the AP happens usually when flashing the esp32 for the first time or some old data is not overwritten. Doing a clear before flashing should solve that issue and takes just a few more seconds.

Also, would it be viable to add that site also to this repository, so everything related is in one place? Not necessary, but would be nice 👀

Anheledir, 2024-06-04:
> I'm not quite sure how to make the link 
> 
> Is their a easy method to do this?

I'm not aware of a direct call for the latest release, but I used some script or backend-code in other projects. For example:

```bash
# Fetch the latest release information
latest_release=$(curl -s https://api.github.com/repos/prusa3d/Prusa-Firmware-ESP32-Cam/releases/latest)

# Extract the download URLs
download_urls=$(echo $latest_release | grep -oP '"browser_download_url": "\K(.*)(?=")')

# Download each asset
for url in $download_urls; do
    wget $url
done
```

Within the for-loop you could filter only the `.bin` files. Not the most elegant solution, but should work 🫣

ALD-Models, 2024-06-04:
> Also, would it be viable to add that site also to this repository, so everything related is in one place? Not necessary, but would be nice 👀

That would be brilliant! 

I've sorted the WIFI issues so it erases the ESP by default as you only really need to use my website once & just in the process of sorting the firmware link.

Many Thanks

ALD-Models, 2024-06-04:
> > I'm not quite sure how to make the link
> > Is their a easy method to do this?
> 
> I'm not aware of a direct call for the latest release, but I used some script or backend-code in other projects. For example:
> 
> ```shell
> # Fetch the latest release information
> latest_release=$(curl -s https://api.github.com/repos/prusa3d/Prusa-Firmware-ESP32-Cam/releases/latest)
> 
> # Extract the download URLs
> download_urls=$(echo $latest_release | grep -oP '"browser_download_url": "\K(.*)(?=")')
> 
> # Download each asset
> for url in $download_urls; do
>     wget $url
> done
> ```
> 
> Within the for-loop you could filter only the `.bin` files. Not the most elegant solution, but should work 🫣

I've managed to get a workflow working so it retrieves and updates the bin files and version code on the manifest. When you release an update i'll just have to start the workflow and it should be all automated from their!

Zero-Circle, 2024-06-17:
What a fantastic update for this project thanks for building this tool. 
I tried to do it myself as I know my way around it with WLED, not a code wizard though. 
This tool is fantastic hopefully this is built into this repo as the default install method. Thanks for making it @ALD-Models.

johnyHV, 2024-06-29:
Hello @ALD-Models I have new challenge. It is possible added multi board compatibility ?

ALD-Models, 2024-06-30:
Hi @johnyHV 

Do you mean like different bin files for the different ESP boards?

That would definitely be possible! Might be able to get it done within a few days.

I've just got an error email from the workflow because the new releases have the bin files in zips now so i need to work on a way to unzip and extract the bin for each of the boards in the workflow. This would hopefully be resolved soon!

Jake

johnyHV, 2024-06-30:
@ALD-Models 

Yes, I means different bin files for the different ESP boards.

Yes, unfortunately, currently the firmware for each board is packed in a ZIP archive. This makes it more organized and easier to identify which file belongs to which board. At the same time, there aren't that many bin files in the release, so the release is more organized.

ALD-Models, 2024-06-30:
Hi @johnyHV 

I would get that sorted

Would the zip file for the individual boards be named the exact same per release?

Just means that the workflow can easily identify the bin files per board

ALD-Models, 2024-06-30:
That should work now!

If anyone can test it and give me feedback that would be really appreciated!

Also just a side note that the repository automatically syncs to the latest release at 00:00 meaning that when an update has been pushed it wont be synced until that time.

kassi, 2024-07-09:
Getting:
Downlading firmware ESP32_PrusaConnectCam.ino.bootloader.bin failed: 404

ALD-Models, 2024-07-09:
> Getting:
> Downlading firmware ESP32_PrusaConnectCam.ino.bootloader.bin failed: 404

Apologies about that. I think I have managed to patch it up however I need to rewrite the workflow to prevent it from downloading the wrong files to the manifest. At the moment I have suspended the workflow from the automation so I can look into it further.

The installer should be working now 🤞

kassi, 2024-07-09:
Thank you. In the meantime I found and downloaded the zip with bin from the release and uploaded them via a web tool. Worked flawlessly.

louisostrowski, 2024-09-26:
Hi johnny, your loader, what pin does it put the DHT22 data receive on?

ALD-Models, 2024-09-26:
> Hi @ALD-Models , your loader, what pin does it put the DHT22 data receive on?

I'm not 100% sure I'm afraid. I'm just using the generic bin files on the latest firmware page. I'll see if I can find a DHT22 in the next few days and will come back to you.