prusa3d/Prusa-Firmware-ESP32-Cam #36: Cam freezes after some time and then restarts

bin101, 2024-05-31:
The problem is most likely to be reproduced in stream mode, but it also occurs in snapshot mode under circumstances unknown to me. The higher the resolution selected, the faster the cam freezes and then restarts by itself. Do I therefore suspect memory problems?

The cam is connected via an external wifi antenna. I have re-soldered the resistor accordingly.

Perhaps my PSRAM is no longer quite right.

Version used: 1.0.3-rc1

Logs:
```
2024-05-31_06-08-35 - Check maximum log file size
2024-05-31_06-08-35 - Log file size: 322 bytes
2024-05-31_06-08-35 - MicroSdCard task. Stack free size: 576 bytes
2024-05-31_06-08-35 - Read ActifeWifiCfgFlag: 10
2024-05-31_06-08-35 - WiFi status: Wifi connected. SSID: XXXXX, BSSID: XXXX, RSSI: -56 dBm, IP: 192.168.2.110, TX power: 19.5dBm
2024-05-31_06-08-35 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-35 - Free RAM: 92792 bytes
2024-05-31_06-08-35 - Free SPIRAM: 4069083 bytes
2024-05-31_06-08-35 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-36 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-36 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-36 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-37 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-37 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-37 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-38 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-38 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-38 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-38 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-38 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-39 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-39 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-39 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-40 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-40 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-40 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-41 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-41 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-41 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-42 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-42 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-42 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-42 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-42 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-43 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-43 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-43 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-44 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-44 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-44 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-44 - Read ActifeWifiCfgFlag: 10
2024-05-31_06-08-44 - WiFiWatchdog task. Stack free size: 856 bytes
2024-05-31_06-08-45 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-45 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-45 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-46 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-46 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-46 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-46 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-46 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-47 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-47 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-47 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-48 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-48 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-48 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-49 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-49 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-49 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-50 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-50 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-50 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-50 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-50 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-51 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-51 - Photo processing task. Stack free size: 4200 bytes
2024-05-31_06-08-51 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-52 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-52 - Task photo processing. Start sending info
2024-05-31_06-08-52 - Task photo processing. Start sending photo
2024-05-31_06-08-52 - Start sending photo to prusaconnect
2024-05-31_06-08-52 - Sending Photo to PrusaConnect, 24768 bytes
2024-05-31_06-08-52 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-53 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-53 - System task. Stack free size: 2480 bytes
2024-05-31_06-08-53 - Connected to server!
2024-05-31_06-08-53 - Sendig photo
2024-05-31_06-08-53 - Photo with EXIF data sent
2024-05-31_06-08-53 - Send done: 24768/24918 bytes
2024-05-31_06-08-53 - Response:
2024-05-31_06-08-53 - HTTP/1.1 200 OK
2024-05-31_06-08-53 - Date: Fri, 31 May 2024 06:08:53 GMT
2024-05-31_06-08-53 - Content-Type: application/json
2024-05-31_06-08-53 - Content-Length: 19
2024-05-31_06-08-53 - Connection: close
2024-05-31_06-08-53 - Strict-Transport-Security: max-age=31536000; includeSubDomains
2024-05-31_06-08-53 - 
2024-05-31_06-08-54 - SerialCfg task. Stack free size: 616 bytes
2024-05-31_06-08-54 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-54 - {"status_code":204}
2024-05-31_06-08-54 - Full response: HTTP/1.1 200 OK
Date: Fri, 31 May 2024 06:08:53 GMT
Content-Type: application/json
Content-Length: 19
Connection: close
Strict-Transport-Security: max-age=31536000; includeSubDomains

{"status_code":204}
2024-05-31_06-08-54 - SystemLed task. Stack free size: 316 bytes
2024-05-31_06-08-54 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
0000-00-00_00-00-00 - Log file size: 328 bytes
----------------------------------------------------------------
Start MCU!
SW Version: 1.0.3-rc1 ,Build: May 27 2024 18:08:00
Verbose mode: false
Log level: 3
```

Settings:
[image]

johnyHV, 2024-05-31:
Hello @bin101 . Yes, this is RAM issue. The current software version 1.0.3-rc2 should already solve these problems

bin101, 2024-06-01:
@johnyHV It got better but still not 100% reliable:
```
2024-06-01_10-41-36 - WiFi status: Wifi connected. SSID: XXX, BSSID: XXX, RSSI: -51 dBm, IP: 192.168.2.110, TX power: 19.5dBm
2024-06-01_10-41-36 - Free RAM: 34484 B
2024-06-01_10-41-36 - Free SPIRAM: 3986963 B
2024-06-01_10-41-36 - Temperature: 92.78 *C
2024-06-01_10-41-36 - WiFiManagement task. Stack free size: 948B
2024-06-01_10-41-36 - WiFi status: 3
2024-06-01_10-41-37 - System task. Stack free size: 684B
2024-06-01_10-41-37 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-38 - System task. Stack free size: 684B
2024-06-01_10-41-38 - SystemLed task. Stack free size: 592B
2024-06-01_10-41-38 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-38 - SystemLed task. Stack free size: 592B
2024-06-01_10-41-39 - System task. Stack free size: 684B
2024-06-01_10-41-39 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-40 - System task. Stack free size: 684B
2024-06-01_10-41-40 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-41 - System task. Stack free size: 684B
2024-06-01_10-41-41 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-42 - System task. Stack free size: 684B
2024-06-01_10-41-42 - StreamTelemetry task. Stack free size: 276B
2024-06-01_10-41-42 - Stream, average data in 30sec. FPS: 0.0, Size: 0KB
2024-06-01_10-41-42 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-42 - SystemLed task. Stack free size: 592B
2024-06-01_10-41-42 - SystemLed task. Stack free size: 528B
2024-06-01_10-41-43 - System task. Stack free size: 684B
2024-06-01_10-41-43 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-44 - System task. Stack free size: 684B
2024-06-01_10-41-44 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-45 - System task. Stack free size: 684B
2024-06-01_10-41-45 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-46 - System task. Stack free size: 684B
2024-06-01_10-41-46 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-46 - SystemLed task. Stack free size: 528B
2024-06-01_10-41-46 - SystemLed task. Stack free size: 528B
2024-06-01_10-41-47 - System task. Stack free size: 684B
2024-06-01_10-41-47 - SerialCfg task. Stack free size: 912B
2024-06-01_10-41-48 - System task. Stack free size: 684B
2024-06-01_10-41-48 - SerialCfg task. Stack free size: 912B
0000-00-00_00-00-00 - Log file size: 368/1024 B
----------------------------------------------------------------
Start MCU!
SW Version: 1.0.3-rc2 ,Build: May 31 2024 22:46:47
Verbose mode: false
Log level: 3
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - CPU reset reason: Reset due to task watchdog
0000-00-00_00-00-00 - CPU0 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - CPU1 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - MCU Temperature: 91.67 *C
0000-00-00_00-00-00 - Internal Total heap: 296896 ,internal Free Heap: 264088
0000-00-00_00-00-00 - SPIRam Total heap: 4192091 ,SPIRam Free Heap: 4165155
0000-00-00_00-00-00 - ChipRevision: 3 ,Cpu Freq: 240 ,SDK Version: v4.4.7-dirty
0000-00-00_00-00-00 - Flash Size: 4194304 ,Flash Speed 80000000
0000-00-00_00-00-00 - PSRAM is used.
0000-00-00_00-00-00 - malloc/new is not using SPIRAM
0000-00-00_00-00-00 - Init cfg module: 391
0000-00-00_00-00-00 - Read FirstMcuStart: 
0000-00-00_00-00-00 - It's not first start MCU: 15
```

johnyHV, 2024-06-01:
Can you send me fuul log ? From MCU start to MCU reboot ? My camera running an hour without a problem.

bin101, 2024-06-01:
@johnyHV send it to your Gmail address

johnyHV, 2024-06-02:
@bin101 I analyzed your logs. Video streaming is generally a complex task, especially when combined with sending photos to the backend. It uses a large amount of RAM and computational CPU performance. In the latest version of the firmware that I just released, the issues you had in the logs should be resolved. Currently, ESP IDF 4.7 is being used. There is a newer version, ESP IDF 5.1, which I works to update to, but minor adjustments to the source code are needed. I anticipate that upgrading to ESP IDF 5.1 will increase the stability and reliability of the device, as I suspect that ESP IDF 4.7 has significant memory leaks that I unfortunately cannot control.

johnyHV, 2024-06-30:
Hello @bin101 I published new release, 1.1.0-rc1. In this version is used new ESP IDF library, and new version of ESPAsyncWebServer. This changes make better performance and stability. Can you tested it please ?

bin101, 2024-06-30:
I will

bin101, 2024-06-30:
@johnyHV uptime seems to be improved without memory related reboots but now I have Wifi issues:

```
Start MCU!
SW Version: 1.1.0-rc1 ,Build: Jun 29 2024 17:32:14
Verbose mode: false
Log level: 1
0000-00-00_00-00-00 - CPU reset reason: Software reset via esp_restart
0000-00-00_00-00-00 - CPU0 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - CPU1 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - Service AP mode disabled!
0000-00-00_00-00-00 - Connecting to WiFi: LordOfThePings_24
2024-06-29_16-55-33 - Photo without EXIF data sent
2024-06-29_17-09-29 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_17-09-29 - WiFi disconnected from access point. Reason: 8
2024-06-29_17-22-33 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_17-22-33 - WiFi disconnected from access point. Reason: 8
2024-06-29_17-59-29 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-06-29_17-59-53 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_17-59-53 - WiFi disconnected from access point. Reason: 8
2024-06-29_18-12-01 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_18-12-01 - WiFi disconnected from access point. Reason: 8
2024-06-29_18-16-41 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_18-16-41 - WiFi disconnected from access point. Reason: 8
2024-06-29_18-35-49 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_18-35-49 - WiFi disconnected from access point. Reason: 8
2024-06-29_18-47-01 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_18-47-01 - WiFi disconnected from access point. Reason: 8
2024-06-29_18-57-45 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_18-57-45 - WiFi disconnected from access point. Reason: 8
2024-06-29_18-57-45 - WiFi WDG. STA connection lost.
2024-06-29_18-58-31 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-06-29_18-59-06 - WiFi disconnected from access point. Reason: 6
2024-06-29_18-59-09 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-09 - Reconnecting to WiFi. STA
2024-06-29_18-59-09 - Disconnect from WiFi
2024-06-29_18-59-09 - Reconnecting to WiFi. STA
2024-06-29_18-59-09 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-06-29_18-59-12 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-15 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-18 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-21 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-24 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-25 - WiFi WDG. STA connection lost.
2024-06-29_18-59-28 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-31 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-34 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-37 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-37 - Reconnecting to WiFi. STA
2024-06-29_18-59-37 - Disconnect from WiFi
2024-06-29_18-59-37 - Reconnecting to WiFi. STA
2024-06-29_18-59-40 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-43 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-45 - WiFi WDG. STA connection lost.
2024-06-29_18-59-46 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-49 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-52 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-55 - WiFi disconnected from access point. Reason: 15
2024-06-29_18-59-58 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-01 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-04 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-05 - Reconnecting to WiFi. STA
2024-06-29_19-00-05 - Disconnect from WiFi
2024-06-29_19-00-05 - Reconnecting to WiFi. STA
2024-06-29_19-00-05 - WiFi WDG. STA connection lost.
2024-06-29_19-00-07 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-10 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-14 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-17 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-00-20 - WiFi disconnected from access point. Reason: 15
[... 38 lines cut ...]
2024-06-29_19-01-39 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-01-43 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-01-45 - WiFi WDG. STA connection lost.
2024-06-29_19-01-46 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-01-49 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-01-52 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-01-55 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-01-57 - Reconnecting to WiFi. STA
2024-06-29_19-01-57 - Disconnect from WiFi
2024-06-29_19-01-57 - Reconnecting to WiFi. STA
2024-06-29_19-01-58 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-01 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-04 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-05 - WiFi WDG. STA connection lost.
2024-06-29_19-02-07 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-10 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-13 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-16 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-19 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-22 - WiFi disconnected from access point. Reason: 15
2024-06-29_19-02-25 - Reconnecting to WiFi. STA
2024-06-29_19-02-25 - Disconnect from WiFi
2024-06-29_19-02-25 - WiFi WDG. STA connection lost.
2024-06-29_19-02-25 - Reconnecting to WiFi. STA
```

johnyHV, 2024-07-02:
@bin101 can you send me please fuul log ?

bin101, 2024-07-03:
@johnyHV send the logs to your gmail address

johnyHV, 2024-07-08:
@bin101 Thank you for the logs. I looked through them, and it looks strange. It might be issues with your internet connection or an unusual error with the library. A new version of ESP-IDF has just been released, so I will try to release rc2 in the near future. Maybe that will help resolve your problem.

bin101, 2024-07-18:
@johnyHV with 1.1.0-rc2 I now reached an uptime of 22hours still counting. Let's see if it stays this way ;)

bin101, 2024-07-26:
@johnyHV I noticed one new condition for reboots:

Every time the Maximum log file size is reached it will restart.

So the log ends with e.g. 
`2024-07-26_05-24-13 - Maximum log file size. File count: 272`
and in the next log I see the reboot:

```
0000-00-00_00-00-00 - Log file size: 0/1024 B
----------------------------------------------------------------
Start MCU!
SW Version: 1.1.0-rc2 ,Build: Jul 12 2024 22:37:52
Verbose mode: false
Log level: 3
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - SW Version: 1.1.0-rc2 Build: Jul 12 2024 22:38:08
0000-00-00_00-00-00 - Board name: AI Thinker ESP32-CAM
0000-00-00_00-00-00 - CPU reset reason: Software reset due to exception/panic
```

johnyHV, 2024-07-31:
Hello @bin101 . Thank you for the report, but I can't simulated same issue. Can you try another micro SD card ? Or Format the micro SD card ?

bin101, 2024-08-01:
Hi @johnyHV did you try excessive logging in verbose mode? For now I switched to a other SD card, maybe this will solve it already.

johnyHV, 2024-08-02:
Hello @bin101 , Yes I tried verbose log level, and nothing. But I had a problem with one SD card. If I had it inserted micro SD card in the camera, the MCU couldn't boot and was stuck in a strange boot loop.

bin101, 2024-08-02:
Hi @johnyHV, send you another log via mail. I think now there is just a wifi issue left which could cause a software reset due to exception/panic but I am not really sure.

Using  a better SD card definitely helped for most of the reboots.

johnyHV, 2024-08-04:
Hello @bin101 . Here is new, experimentaly FW. Can you test it please ? 

[esp32-cam.zip](https://github.com/user-attachments/files/16487685/esp32-cam.zip)

bin101, 2024-08-08:
@johnyHV Cannot see the wifi issue so far anymore, but random reboots without any hints, send you the logs anyway.

johnyHV, 2024-08-22:
Hello @bin101, could you please attach the logs here? Please ensure that all sensitive content, such as passwords and tokens, is hidded. It will help me navigate through the issues more easily. Thank you very much

bin101, 2024-08-27:
@johnyHV Sure here it is, but maybe then flag specific logs as sensible in the future? For an easy export?:

[SysLog.log2.txt](https://github.com/user-attachments/files/16758240/SysLog.log2.txt)
[SysLog.log10.txt](https://github.com/user-attachments/files/16758242/SysLog.log10.txt)
[SysLog1.log1.txt](https://github.com/user-attachments/files/16758243/SysLog1.log1.txt)

bin101, 2024-10-04:
@johnyHV any findings?

johnyHV, 2025-01-03:
Hello @bin101 sorry for the late reply. You still have same issue with latest FW ?

bin101, 2025-01-06:
Haven't had a reboot in 2 days. Closing this now.

closed 2025-01-06 by bin101