prusa3d/Prusa-Firmware-ESP32-Cam #38: ESP32 flash Hz bar artefacts

Simon7381, 2024-06-01:
There seems to be a Hz interference with the LED used for the flash.
Please note, I have rotated the image 270d, so these are horizontal banding lines with the print bed to the right and extruder at the bottom.

- With the LED off and illuminated by room LED lights:
- - [image]
- With the LED on constantly and illuminated by room LED lights:
- - [image]
- With the LED on flash and illuminated by room LED lights:
- - [image]
- With the LED on flash:
- - [image]
- With the LED on flash, battery USB:
- - [image]
- Original Prusa Enclosure LED strip:
- - [image]

I'm in the UK with 50Hz electricity, with ESP32 from https://amzn.eu/d/54PickP
I've tried different USB power supplies, including a battery to eliminate the 50Hz mains frequency.

I guess this is the LED on the board, but is it I've got a bad board or is this some LED driver pulsing the flash, is there any setting that can be done to eliminate or tune out this interference banding?

johnyHV, 2024-06-02:
Hello @Simon7381. I am in the EU and also have 50Hz electricity. I haven't seen this problem yet. Currently, you need to manually change the setting in the code, in the mcu_cfg.h file, and variable

`
#define FLASH_ON_STATUS   205   ///< PWM intensity LED for ON. limitation to 80%. 2^FLASH_PWM_RESOLUTION * 0.8% = 204
#define FLASH_PWM_FREQ   2000               ///< frequency of pwm [240MHz / (100 prescale * pwm cycles)] = frequency
`

In the future I can added new parameter on the WEB page, for settings PWM intensity. Can you test change at first step FLASH_ON_STATUS    ?

Simon7381, 2024-06-02:
> Hello @Simon7381. I am in the EU and also have 50Hz electricity. I haven't seen this problem yet. Currently, you need to manually change the setting in the code, in the mcu_cfg.h file, and [...]

I had been lazzy and used the binnary to upload, so will give me an excuse to get the arduino setup on Monday 😀
Will let you know, thanks

johnyHV, 2024-06-13:
Hi @Simon7381 . LED diode on the PCB don't have resistor for the current limitation. So currently is for testing changed PWM to 80%. Maybe this "solution" should be help extend LED diode life time. Can you try use original PWM frequency, and 100% PWM ?

closed 2024-06-27 by johnyHV