prusa3d/Prusa-Firmware-ESP32-Cam #39: Image Rotation not applied to stream.mjpg

ice987987, 2024-06-02:
the new `image rotation` feature, implemented with `1.0.3-rc2`, is not applied to the stream `http://[ip]/stream.mjpg`.

johnyHV, 2024-06-02:
@ice987987 Yes, the stream currently does not support EXIF metadata. I will try to implemented it in the next version of the firmware. For now, I don't know if it will be possible to use EXIF metadata with the stream. EXIF metadata must be calculated for each image individually. Adding EXIF metadata may cause higher CPU load and may reduce the number of FPS. These are just estimates, and I need to test it.

twjdeboer, 2024-06-10:
Visual indication of the problem: The small thumbnails do not rotate according to the setting, but the camera view does.

[image]

johnyHV, 2024-06-30:
@ice987987 Experimentally, I added EXIF for the stream, but so far I am not satisfied with the board's performance. By default, EXIF for the stream is turned off. However, you can enable it by using the macro **CAMERA_EXIF_ROTATION_STREAM** in **mcu_cfg.h**.

ice987987, 2024-06-30:
thank you very much!

closed 2024-06-30 by ice987987