prusa3d/Prusa-Firmware-ESP32-Cam #41: URL for control Light

Flavien06, 2024-06-03:
Hi
Would it be possible to add a simple URL request,
for example, http://prusa-esp32cam.local/light_on and http://prusa-esp32cam.local/light_off (or other solution)
to control the LED from a home automation application (to activate it automatically at night, for example)?

Thank

blurfl, 2024-06-03:
This approach makes more sense than a control through PrusaConnect or PrusaLink.

johnyHV, 2024-06-04:
Hello @Flavien06 . This is good idea. I can added requests

johnyHV, 2024-06-30:
Hello @Flavien06 WEB API is already added [here](https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam?tab=readme-ov-file#web-api)

Flavien06, 2024-07-01:
OK nice. I juste udate to v1.1.0-rc1

All commande works exept light and flash

For http://ip/light?on 
```
URL not Found

URI: /light
Method: GET
Arguments: 1
 on: 
```

johnyHV, 2024-07-08:
@Flavien06 for me it works. Please, can you try it again ? 

[image]

Flavien06, 2024-07-19:
I tried but still url not found. I'll try to reflash the firmware later.

Flavien06, 2024-07-19:
It's good after new flashing of firmware it works.

_Thanks :) I added the functions to my script
https://github.com/Flavien06/PrusaLinkAPI-SH

Flavien06, 2024-07-19:
It's good after new flashing of firmware it works.

_Thanks :) I added the functions to my script
https://github.com/Flavien06/PrusaLinkAPI-SH_

closed 2024-07-19 by Flavien06