prusa3d/Prusa-Firmware-ESP32-Cam #42: ESP32-CAM Locking up

janetpoague, 2024-06-05:
First let me start by letting everyone know I am novice with the ESP32-CAM.  Purchased 2 from Amazon.  Espressif used to install firmware.   Been experiencing lock ups.   Several times had to reflash the firmware as no service ap started after power cycle.   I can't quite put my finger on the specifics to make it die every time using same steps.   Seems to be this scenario.  Connect to the camera config page.  Ping to service ip 192.168.0.1 -t.  This is to watch when cam dies.  Click start stream which starts the stream.   Go back to camera config and click scan wi-fi for example.   Camera loses pings.   Happens in probably less then 2 min.   Sometimes it drops 3 - 6 then reconnects but other times its dead.   Rebooting camera by pulling usb cable and reconnecting does not bring it back in all cases.   In this situation flashing firmware resurrects it.  Over past 10 days probably flashed firmware 10 times.   I am currently on 1.0.3.  Few screen shots below of settings.  Have experienced same with 2 other firmware version.  Ideas recommendations?

[image]

[image]

janetpoague, 2024-06-29:
> Hello @janetpoague Is this problem still persisting?

Yes.  Still having the issue. Janet Poague

johnyHV, 2024-06-30:
@janetpoague Can you try new FW version ? And send me logs from devices ?

janetpoague, 2024-07-02:
Updated firmware yesterday.    Had issues on my side.  Did not realize when
prusa xl when in an offline state they don't update the camera pics in
connect.  Had to reformat the memory card for the logs too.  It all of a
sudden said it was not inserted.    Possibly it was like that post fw
upgrade and did not notice.  Card being detected after reformat. Once I got
passed that seemed to be working ok for a couple of hours.  Was not getting
any lock ups (meaning going to different pages - camera - wifi -
authentication - system) on  the camera config page (192.168.0.1) until I
clicked "start stream".  Probably should have left well enough alone.  The
stream worked for maybe less than a minute then intermittent ping drops to
192.68.0.1 and navigation between camera - wifi - authentication - system
did not work.  Page mostly locked up. Rebooted a few times.  Now I'm
getting the error below.  (partial log included too)   No idea what it
means.  Have power cycled camera probably 1/2 dozen times today.  The
strange lock up behavior after clicking the start stream happened in the
prior fw version too.  Only way that seemed to stabilize it was to reload
firmware.   Going to let this one sit all night and see what it does
tomorrow.

I have another camera.  Same model.  Going to put the new firmware
upgrade on it shortly.  I will not click the start stream and see how it
behaves.  Can report back in a later email.

Connection failed to domain! Error: -32512 - SSL - Memory allocation failed
: 0 ,BA:Backend unavailable

2024-07-02_22-00-58 - Taking photo...
2024-07-02_22-00-58 - The picture has been saved. Size: 49326 bytes,
Photo resolution: 1024 x 768
2024-07-02_22-00-58 - Photo OK! 0
2024-07-02_22-00-58 - Exif header OK! Len: 310
2024-07-02_22-00-58 - Start sending photo to prusaconnect
2024-07-02_22-00-58 - Sending Photo to PrusaConnect, 49616 bytes
2024-07-02_22-00-58 - Connetion failed to domain! Error: -32512 - SSL
- Memory allocation failed : 0 ,BA:Backend unavailable
2024-07-02_22-01-00 - WiFi status: Wifi connected. SSID: LBF-Ops-2.4,
BSSID: FC:7F:F1:80:B6:C0, RSSI: -65 dBm, IP: 192.168.110.102, TX
power: 19.5dBm
2024-07-02_22-01-00 - Reconnecting to WiFi. STA. Problem with
connecting to backend!
2024-07-02_22-01-00 - WiFi disconnected from access point. Reason: 8
2024-07-02_22-01-00 - Free RAM: 101288 B, Min: 33160
2024-07-02_22-01-00 - Free PSRAM: 4007784 B, Min: 3947704
2024-07-02_22-01-00 - Temperature: 53.33 *C
2024-07-02_22-01-00 - WiFi connected to STA
2024-07-02_22-01-00 - WiFi Got IP address: 192.168.110.102
2024-07-02_22-01-00 - WiFi Got mask: 255.255.254.0
2024-07-02_22-01-00 - WiFi Got gateway: 192.168.111.1
2024-07-02_22-01-00 - WiFi Got DNS 1: 192.168.11.10
2024-07-02_22-01-00 - WiFi Got DNS 2: 10.150.150.148
2024-07-02_22-01-09 - Start sending device information to prusaconnect
2024-07-02_22-01-09 - Data:
{"config":{"name":"xl","resolution":{"width":1024,"height":768},"network_info":{"wifi_mac":"FC:E8:C0:CE:18:84","wifi_ipv4":"192.168.110.102","wifi_ssid":"LBF-Ops-2.4"}}}
2024-07-02_22-01-09 - Sending Info to PrusaConnect, 169 bytes
2024-07-02_22-01-09 - Connetion failed to domain! Error: -32512 - SSL
- Memory allocation failed : 0 ,BA:Wait for first connection
2024-07-02_22-01-09 - Taking photo...
2024-07-02_22-01-09 - The picture has been saved. Size: 49357 bytes,
Photo resolution: 1024 x 768
2024-07-02_22-01-09 - Photo OK! 0
2024-07-02_22-01-09 - Exif header OK! Len: 310
2024-07-02_22-01-09 - Start sending photo to prusaconnect
2024-07-02_22-01-09 - Sending Photo to PrusaConnect, 49647 bytes
2024-07-02_22-01-09 - Connetion failed to domain! Error: -32512 - SSL
- Memory allocation failed : 0 ,BA:Wait for first connection
2024-07-02_22-01-19 - CardStatus: 1 FileStatus: 1
2024-07-02_22-01-20 - Start sending device information to prusaconnect
2024-07-02_22-01-20 - Data:
{"config":{"name":"xl","resolution":{"width":1024,"height":768},"network_info":{"wifi_mac":"FC:E8:C0:CE:18:84","wifi_ipv4":"192.168.110.102","wifi_ssid":"LBF-Ops-2.4"}}}
2024-07-02_22-01-20 - Sending Info to PrusaConnect, 169 bytes
2024-07-02_22-01-20 - Connetion failed to domain! Error: -32512 - SSL
- Memory allocation failed : 0 ,BA:Wait for first connection

Janet Poague

johnyHV, 2024-08-22:
Hello @janetpoague . Yes, stream funcionality is too complicated in the FW. The stream is using too many hardware resources, and sometimes the RAM can run out when uploading photos to PrusaConnect while simultaneously sending photos via the stream. Are you still experiencing this issue?

muesli, 2024-10-20:
Maybe the uploading to PrusaConnect could be paused while an mjpeg stream is active?

johnyHV, 2024-11-20:
@muesli yes, this is one option how to solved this issue.

Morcegolas, 2024-11-20:
Hello, I've this exact problem. I bought 5 cameras on AliExpress at least one year ago, and had this problem on all of them, thought it was a problem with the cheap cameras and moved on. Yesterday I bought a new camera with WiFi antena from Amazon, exact same problem. I'm using my Mac and Arduino IDE to flash it. 
I'm only adding my WiFi name and pass, add the camera token, and change the resolution to 800*600 or 1024*768 and nothing else. 
I cannot understand what I'm missing here :/

Morcegolas, 2024-11-20:
Hello, I've this problem but I don't activate streaming. I bought 5 cameras on AliExpress at least one year ago, and had this problem on all of them, thought it was a problem with the cheap cameras and moved on. Yesterday I bought a new camera with WiFi antena from Amazon, exact same problem. I'm using my Mac and Arduino IDE to flash it. 
I'm only adding my WiFi name and pass, add the camera token, and change the resolution to 800*600 or 1024*768 and nothing else. 
I cannot understand what I'm missing here :/

randall3000, 2024-12-01:
I had the same problem, I just max the trigger interval and problem solved, stream is solid now, no frezee anymore.

dabijja, 2025-02-12:
I've had the same issue with this camera model and I have found that if I am not using the MB module but wiring 5v and GND directly to the pins of the esp (after flashing it etc) from a regular mobile phone charger, most of my issues disappeared.

randall3000, 2025-02-12:
@dabijja  thanks for the tip, I'll try later today.

itszeekay, 2025-10-17:
I have the same issue with the same units as @djaesthetic I noticed that on direct wifi, the configuration works just fine. On client WiFi, accessing the UI and changing settings (sometimes) or viewing the stream causes the esp to basically crash. If I just power cycle it, it comes back online and shows in prusa connect and is otherwise stable.