prusa3d/Prusa-Firmware-ESP32-Cam #44: Wifi only works when connected to computer USB

esserc, 2024-06-14:
I'm not sure if this is purely a problem with Prusa-Firmware-ESP32-Cam or not. The digging I've been able to do seems to indicate that the unit I have only works properly when connected to a USB serial device. I'm not experienced enough with the ESP32 to really understand why.

johnyHV, 2024-06-17:
This issue looks as problem with power supply, or programmer. Can you try connect 5V power supply to camera module without programmer board ? Or can you are use different power supply with minimal 5V and 2A ?

esserc, 2024-06-17:
I tried it with a 3.1A power supply through the programmer and it failed, but when I connected it directly to 5V without the programmer it works perfectly. 

It looks like my specific programmer board needs to see a serial console or it hangs. I'm going to try to find a USB board without a programmer, or a different programmer to see if that solves it.

Thanks!

johnyHV, 2024-06-27:
@esserc Can you try connecto board to power supply, and then reset the board ? Maybe you have problem with the BOOT button. Some people wrote me message and mentioned problems with the BOOT button. Solution is soldering 100nF capacitor to BOOT button

MKornprobst, 2025-10-15:
Hey, so i had the same problem.
The camera would work normally with an external power supply or i had to press the reset/boot button while connected to the printer.
What worked for me was disabling the Access Point for the camera in the web GUI.