prusa3d/Prusa-Firmware-ESP32-Cam #45: WebDAV support

hickengr, 2024-06-16:
I would like to save my snapshots locally to the SD card, retrieve them by copying them via MacOS file manager or Windows file explorer via my local network, and then delete them from the SD card to make room for the next print snapshots.  This could be done via WebDAV.  Is it possible to add WebDAV file transfer?

johnyHV, 2024-06-17:
I'm not sure if it's possible. Adding the webdavn service could slow down the processor and make it unstable. It is another service that consumes the processor and RAM. I'll try to look into it

mozgy, 2024-06-17:
> I'm not sure if it's possible. Adding the webdavn service could slow down the processor and make it unstable. It is another service that consumes the processor and RAM. I'll try to look into it

try something like I did in my [ESP32-Cam](https://github.com/mozgy/esp32-cam), with dir alike hierarchy on URL/archive ..

hickengr, 2024-07-04:
I tried getting the ESP32-Cam software at link to work with little success.  For some reason, the SD card is not getting recognized and the settings page is not found.  I tried copying the web interface to the SD card in the location specified and the software just does not find the settings page.  Info page is found with no problems.  But I think the infor page is embedded into the firmware unlike the settings page.  I am not a software engineer, but I have been writing software for decades.  Maybe I am not understanding the web server software correctly. 

> try something like I did in my [ESP32-Cam](https://github.com/mozgy/esp32-cam), with dir alike hierarchy on URL/archive ..

mozgy, 2024-07-04:
> I tried getting the ESP32-Cam software at link to work with little success. For some reason, the SD card is not getting recognized and the settings page is not found. I tried copying the web [...]

Sounds like you flashed only code part of firmware, you need data part as well .. Open an issue there so we can discuss more ..