prusa3d/Prusa-Firmware-ESP32-Cam #46: Disconnected Camera results in boot loop.

Zero-Circle, 2024-06-17:
If the camera is not connected correctly or totally disconnected this results in a boot loop.

IMO if camera is not detected it should continue booting and throw an error on the website.
The device already knows that camera is disconnected it should be allowed to continue through this error.

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
----------------------------------------------------------------
Start MCU!
Prusa ESP32-cam https://prusa3d.cz
SW Version: 1.0.3
Build: Jun  2 2024 21:20:32
----------------------------------------------------------------
Init Logs library
Start init micro-SD Card
E (283) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (284) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
SD Card Mount Failed
Micro-SD card not found! Disable logs
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - CPU reset reason: Reset due to power-on event

** Removed section of assumed irrelevent boot info**

0000-00-00_00-00-00 - Init GPIO
**0000-00-00_00-00-00 - Init camera module
E (8057) sccb: SCCB_Write Failed addr:0x30, reg:0x3e, data:0x00, ret:263
E (8058) camera: Camera probe failed with error 0xffffffff(ESP_FAIL)
0000-00-00_00-00-00 - Camera init failed. Error: ffffffff
0000-00-00_00-00-00 - Reset ESP32-cam!**
0000-00-00_00-00-00 - WiFi STA stop
ets Jul 29 2019 12:21:46
`

johnyHV, 2024-06-17:
It has happened to me many times that if the camera is not detected, restarting the MCU helps. Sometimes this can occur during the camera restart using WDG, and after subsequently turning on the software, the camera is not detected. A subsequent restart of the MCU causes the camera to be found and functional.

Therefore, removing the reset could cause the camera to not be detected at times. This could lead to worse complications and problems.

Zero-Circle, 2024-06-18:
This makes sense, I was kinda thinking it couldnt have been an oversight.

My problem was from frustration as I was stuck in a boot loop no matter how many times I reseated the camera. As soon as anew camera arrived and I tried it, the system came up as expected. I am assuming that I have a dead camera.

Is there anyother diagnosis tools that the mcu can perform to diagnose the camera other than a binary yes or no?

mozgy, 2024-06-18:
I had once camera misbehaving like that, cleaning contacts with isopropyl helped ..