prusa3d/Prusa-Firmware-ESP32-Cam #47: Wifi STA Appears to connect, but disconnects repeatedly

wlockwood, 2024-06-19:
I upgraded this morning in the hopes of curing the random freezing with an old version of this firmware but now wifi is now effectively not functional.

ESP32 appears to boot normally, then - if I'm reading the serial output right - tries to run services for both STA and AP concurrently, then constantly gets booted from wifi for "reason 2", which is apparently expired auth. I'm not using RADIUS or anything crazy.

I've got it running via an external antenna and it's about ten feet from a wireless access point.

Serial output:
```
0000-00-00_00-00-00 - mDNS: prusa-esp32cam
0000-00-00_00-00-00 - Enable Service AP: 1
0000-00-00_00-00-00 - Network IP method: DHCP
0000-00-00_00-00-00 - Network static IP: 255.255.255.255
0000-00-00_00-00-00 - Network static mask: 255.255.255.255
0000-00-00_00-00-00 - Network static gateway: 255.255.255.255
0000-00-00_00-00-00 - Network static DNS: 255.255.255.255
0000-00-00_00-00-00 - WiFi MAC: C8:2E:18:25:1B:94
0000-00-00_00-00-00 - Service AP SSID: ESP32_camera_37.27.148
0000-00-00_00-00-00 - Service AP mode enabled
0000-00-00_00-00-00 - WiFi STA start
0000-00-00_00-00-00 - WiFi AP start
0000-00-00_00-00-00 - Service IP Address: http://192.168.0.1
0000-00-00_00-00-00 - STA IP Method: DHCP
0000-00-00_00-00-00 - Scan WI-FI networks
0000-00-00_00-00-00 - Check available WI-FI network: myssid
0000-00-00_00-00-00 - WiFi networks scan done
0000-00-00_00-00-00 - 9 networks found
0000-00-00_00-00-00 - Nr | SSID                             | RSSI | CH | BSSID             | Encryption
0000-00-00_00-00-00 -  5 | myssid                           |  -64 | 11 | 52:22:54:37:8B:5E | WPA+WPA2
0000-00-00_00-00-00 -  8 | myssid                           |  -84 |  1 | 26:61:B4:CC:6B:6C | WPA+WPA2
0000-00-00_00-00-00 - SSID: myssid found, 2x
0000-00-00_00-00-00 - WiFi roaming found! Connecting to 52:22:54:37:8B:5E -> -64dBm, 1
0000-00-00_00-00-00 - Connecting to STA BSSID
0000-00-00_00-00-00 - Connecting to WiFi: myssid
0000-00-00_00-00-00 - Starting mDNS record: http://prusa-esp32cam.local
0000-00-00_00-00-00 - Starting mDNS OK
0000-00-00_00-00-00 - Init camera lib
0000-00-00_00-00-00 - Init GPIO
0000-00-00_00-00-00 - Init camera module
0000-00-00_00-00-00 - Set camera CFG
0000-00-00_00-00-00 - Get camera model and type
0000-00-00_00-00-00 - Camera type: 38
0000-00-00_00-00-00 - Camera name: OV2640
0000-00-00_00-00-00 - Camera model: 1
0000-00-00_00-00-00 - Camera PID: 38
0000-00-00_00-00-00 - Camera MAX framesize: 13
0000-00-00_00-00-00 - Camera support jpeg: 1
0000-00-00_00-00-00 - Taking photo...
0000-00-00_00-00-00 - The picture has been saved. Size: 11605 bytes, Photo resolution: 320 x 240
0000-00-00_00-00-00 - Photo OK! 0
0000-00-00_00-00-00 - Exif header OK! Len: 310
0000-00-00_00-00-00 - Starting init WEB server
0000-00-00_00-00-00 - Init PrusaConnect lib
0000-00-00_00-00-00 - Start tasks
0000-00-00_00-00-00 - System task. core: 0
0000-00-00_00-00-00 - SerialCg task. core: 0
0000-00-00_00-00-00 - StreamTelemetry task. core: 0
0000-00-00_00-00-00 - WiFiWatchdog task. core: 0
0000-00-00_00-00-00 - SystemLed task. core: 0
0000-00-00_00-00-00 - Init WDG
0000-00-00_00-00-00 - MicroSdCard check task. core: 0
0000-00-00_00-00-00 - CardStatus: 0 FileStatus: 0
0000-00-00_00-00-00 - Task Wifi Management. core: 0
0000-00-00_00-00-00 - MCU configuration done
0000-00-00_00-00-00 - Free RAM: 85052 B
0000-00-00_00-00-00 - Free SPIRAM: 4176415 B
0000-00-00_00-00-00 - Temperature: 72.78 *C
0000-00-00_00-00-00 - Task photo processing. core: 0
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - CardStatus: 0 FileStatus: 0
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - Free RAM: 85084 B
0000-00-00_00-00-00 - Free SPIRAM: 4176415 B
0000-00-00_00-00-00 - Temperature: 73.89 *C
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 2
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 204
```

johnyHV, 2024-06-27:
Hello @wlockwood Is this problem still persisting? It might be necessary to erase the FLASH memory in the MCU and set it up again.

wlockwood, 2024-06-28:
Flashing solved this issue... though now the web interface doesn't respond.

closed 2024-06-28 by wlockwood