prusa3d/Prusa-Firmware-ESP32-Cam #48: Incorrectly labeled pin for LED relay

Zero-Circle, 2024-06-23:
I started diagnosing why my relay wasnt firing when turning the flash on and off and discovered that the relay schematic in [this section](https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam?tab=readme-ov-file#flash-led-issue) is incorrect.

On the ESP32cam schmatic it is labelled as HS2_Data_1, in the above diagram it says to connect it to HS2_CLK.
[image]
[image]

The diagram needs to be updated to look like this(red line indicates new route):
[image]

If I've somehow missed something here please let me know. This is a working config on my all my boards so I dont think I have a dud board or different board revision.

johnyHV, 2024-06-27:
@Zero-Circle thanks i'll check it out

johnyHV, 2024-06-27:
@Zero-Circle yes, you have right. Thank you for the correction

closed 2024-06-27 by johnyHV