prusa3d/Prusa-Firmware-ESP32-Cam #49: Add temp display in settings menu

Zero-Circle, 2024-06-23:
It would be nice to have a temp readout in the settings menu.

johnyHV, 2024-06-27:
Hello @Zero-Circle you mean MCU temperature, or temperature from external sensor ? I thought about adding support for DHT22, for monitoring the temperature in the box

Zero-Circle, 2024-06-27:
Having both would be good. I think im having an issue due to MCU temp atm and having it in the GUI would be nice.
If you consider the DHT22 it would be nice to expose its value somehow so we can log it in home assistant

johnyHV, 2024-06-27:
@Zero-Circle In the next release, I will add MCU temperature display. I will try to add support for the DHT22. I need to find out how many kilobytes of FLASH memory the support for DHT22/DHT11 will take up.

dc740, 2024-06-30:
Any chance the temperature is in Fahrenheit? My module is the cheap ESP32-CAM:
https://es.aliexpress.com/item/1005006752792592.html
I have two and they are both reading between 101 and 104°C when they have just been powered on, from 98 to 100°C after a few minutes (when the CPU is not being hit that much). They are both just warm to the touch. I need to take the actual temperature, but I can tell they are not 100°C since they are not burning hot.

There is another reason I believe they are not that hot:

1. I'm NOT using the module's AMS1117 power regulator. I'm supplying them 3.3v directly from a buck converter (DSN Mini 360) and four filter capacitors (470uF, 33uF, 0.1uF and 33pF).
2. The boards are also NOT in a case, they are in an open mount: https://www.printables.com/model/134200-esp32-cam-mount 
3. The light is NOT turned on (it only turns on to take a snapshot for a sec every 30 seconds).

In any case I will reply again when I take some actual temperature readings

johnyHV, 2024-07-08:
Hello @dc740 . No, themperature is in the celsius. Maybe you have corrupted internal temperature sensor. The temperature is read through the ESP32 library. I will try to check the code, but it works correctly for me.

I have maybe 6 ESP32-cam module for testing. My camera boards have averate MCU temperature from 70-80°C

mozgy, 2024-07-08:
Per Espressif documentation internal temp sensor on ESP32-S is not supported. It still is there but it's not accurate.

johnyHV, 2024-07-12:
@Zero-Circle I created now new release, 1.1.0-rc2, and this FW version contain support for the external temperature sensor

kleiner1024, 2024-07-22:
And it would be nice to see this temperature from an external sensort (box temperature) in prusa connect.
Maybe it ist possible to print the temperature into the snapshot, Than the informartion ist directly in prusa connect visible, without use of a data field.

johnyHV, 2024-07-31:
Sorry @kleiner1024 but it is not possible to add temperature to the photo. Maybe in the future developers from Prusa create API in the  Prusa connect for sending temperature data

closed 2025-01-03 by johnyHV