prusa3d/Prusa-Firmware-ESP32-Cam #51: Disable cam when not printing

donmaro, 2024-07-16:
Hi, 

is their a setting or config to disable the cam when the printer state is set to finished or idle or something like that?

Have is working now and it is awesome! Just don't want to send photo's 24/7.

gr Marc

mikromcz, 2024-07-17:
I know it's not an answer you are looking for but let me tell you my setup.
The printer is connected through power metering plug. So I have a power consumption overview, when power drops, my smart home sends me a notification that print is finished and, it's also used to control another plug where the light and camera are connected.
So when power increases from 0 to 5W, both light and camera are powered on, and when I turn off the printer both goes automatically off.
If you don't have a smart home, you could use master-slave plugs on the UPS to achieve the same.

donmaro, 2024-07-17:
Thnx for the idea. That could be a nice workaround. I've homeassistant in place. But currently i'm powering the cam from the printer itself so i should work around that. Thnx for the answere.

closed 2024-07-17 by donmaro