prusa3d/Prusa-Firmware-ESP32-Cam #52: HELP.... FOV and focus

donmaro, 2024-07-17:
Hi, Currently this is my image quality:

[image]

This is not that great. how can i adjust the FOV and and the focus to get better quality?

Thanks in advance.

mikromcz, 2024-07-17:
It looks out of the focus. You can twist the lense a bit clockwise or counterclockwise to get the best results. You can print a small wrench like this https://www.printables.com/model/700239-esp32-cam-ov2640-focus-adjustment-wrench-tool or just use pliers 😏

donmaro, 2024-07-18:
hmmm got it in focus but now it is really zoomed in somehow....

[image]

I would like a more zoomed out view

johnyHV, 2024-07-18:
Hello @donmaro camera don't have support for the automatically focus. Here is simple manual with tool for the focus

https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam?tab=readme-ov-file#man_focus

Unfortunately, you have to focus the camera manually

closed 2024-07-30 by johnyHV