prusa3d/Prusa-Firmware-ESP32-Cam #54: WiFi access with no SSID password

aarpi3, 2024-07-30:
Hi,

In our printer lab there is only a guest WiFi access with no password. It's only a captive portal requiring you to agree to the terms and conditions or pre-approved units based on MAC filtering. But trying to connect the camera to this WiFi SSID (with no password) I just get "Save cfg NOT ok!".

I don’t know if it is a feature or bug, but it looks like I'm not able to save (and connect) the SSID without a password. If it’s not a security issue, would it be possible to change this behavior? (I’ve have not tried to compile anything myself). 

BTW, this has also been an issue directly on our printers (MK4) using wifi, but for them I’ve just switched to a cable connection.

johnyHV, 2024-07-30:
Hello @aarpi3 . Yes, this is standard situation. Camera SW must have a WiFi network with encryption.

aarpi3, 2024-07-31:
> Hello @aarpi3 . Yes, this is standard situation. Camera SW must have a WiFi network with encryption.

Thanks for the feedback @johnyHV.

johnyHV, 2024-07-31:
@aarpi3 I can added precompilation macro, for disable check WiFi password

aarpi3, 2024-07-31:
I'm not 100% sure what that involves or if it's alot of work. But I would be happy if that makes it possible to connect to the open WiFi.

johnyHV, 2024-07-31:
I created first tested version now :) . Can you try tested it please ? it is necessary to set the SSID and leave the password blank

[ESP32_PrusaConnectCam.ino.bin.zip](https://github.com/user-attachments/files/16446917/ESP32_PrusaConnectCam.ino.bin.zip)

aarpi3, 2024-08-01:
Thanks, I'll will hopefully manage to test it tonight.

aarpi3, 2024-08-01:
I can confirm that this version connects to wifi without password. But it feels, maybe unrelated, more sluggish.

johnyHV, 2024-08-02:
Hello @aarpi3 Perfect, thanks. But in the official version of the firmware, I cannot disable the WiFi network protection with a password. So if you need to upload a newer firmware, you will have to compile it yourself. Or send me a message, and I will compile the firmware for you.

The speed of communication will not affect the disabling of password protection.

BE-Automation, 2024-08-09:
Hey @johnyHV, alot of companies block these kind of devices on the main network so we too need to be able to connect to wifi networks without passwords that the netwokr itself blocks devices not on a pre-approved list. So it would be great to see this version supported in future releases.

BE-Automation, 2024-08-12:
Hey @johnyHV, I can't get this version to work for me, doesn't seem to boot up a wifi network to connect to. I'm not sure what firmware it is and how to check. Any help would be great!

BE-Automation, 2024-08-12:
Hi @johnyHV are you able to create another one for the latest firmware? That would be amazing!

aarpi3, 2024-08-21:
> I created first tested version now :) . Can you try tested it please ? it is necessary to set the SSID and leave the password blank
> 
> [ESP32_PrusaConnectCam.ino.bin.zip](https://github.com/user-attachments/files/16446917/ESP32_PrusaConnectCam.ino.bin.zip)

Hi @johnyHV. Would this still work with 1.1.0?

BE-Automation, 2024-08-21:
@aarpi3  I tried to use it, but I couldn't get it to work. But I'm fully aware that could be just me implamenting it incorrectly.

johnyHV, 2024-08-22:
Hello @aarpi3  and @BE-Automation. Sorry for the long response... 

In the **mcu_cfg.h** is macro 

```
#define WIFI_DISABLE_UNENCRYPTED_STA_PASS_CHECK true
```

true option is for disable password checking. In the attachmed is new firmware for the 1.1.0. I working on the new FW version, but this is latest stable FW

[PrusaCam 1.1.0-withouth wifi pass.zip](https://github.com/user-attachments/files/16719101/PrusaCam.1.1.0-withouth.wifi.pass.zip)

BE-Automation, 2024-09-11:
That worked! Thanks very much!