prusa3d/Prusa-Firmware-ESP32-Cam #55: So there is no way to get this working on an ESP32-S3-Wroom Freenove board right now? (OTG, UART)

CodeBradley, 2024-08-01:
Flashed the ESP32-S3 prebuilt firmware and it's not broadcasting wifi like my Xiao board is. I'm also having a really hard problem compiling the repo, I think it's related to version numbers and platformio.

johnyHV, 2024-08-02:
Hello @CodeBradley . but ESP32-S3-Wroom Freenove board is not supported now

appcoders, 2024-10-04:
@johnyHV Anything we can help with? Also sitting here with a ESP-S3-Wroom board.

lptheriault, 2024-11-13:
Same situation here, waiting on this device to be supported.

johnyHV, 2025-01-02:
Hello guys . I finally added support for board Freenove ESP32-S3 Wroom!

closed 2025-01-03 by johnyHV

CodeBradley, 2025-01-09:
> Hello guys . I finally added support for board Freenove ESP32-S3 Wroom!

Heyyy, your the man! Thanks. I printed out an arm holder case for this already that fit great but it didn't have the double USB port slots and this didn't support it, time to bust out the Drimmel lol. I might modify the STLs or create my own if there's not a better one out. If anyone reading the comments knows if a solid case setup or can't find anything and want me to create a good one and upload it to printables then please let me know. I need a new Fusion project, but not going to waste time if there's no demand for one.