prusa3d/Prusa-Firmware-ESP32-Cam #56: Struggling with ESP32-CAM flashing

K5OZ, 2024-08-03:
Hello gurus'
I am having troubles using the Esppressif Flash Download Tool.  I think I set up everything per Miroslav's PDF step by step file; however, all I get when trying to perform the Erase Step is an error.  See attached for the error and setup.
Thanks,
Dave
[image]

johnyHV, 2024-08-03:
Hello @K5OZ . You are selected MCU ESP8266 and not ESP32. It's necessary use ESP32 MCU type in the ESP32 FLASH tool.

[image]

K5OZ, 2024-08-03:
Miroslav,Thanks for the quick response!   Yes, I selected ESP32 for the MCU.Sent from my iPhone

how.to.flash.chip.select.jpg (view on web)

johnyHV, 2024-08-03:
No. In the log is selected ESP8266

[image]

K5OZ, 2024-08-04:
Miroslav,Found the issue; I had multiple windows of the flashing tool open and didn’t notice them. When I went to flash it was an old window that had the wrong setup in it.  Killed all programs and started anew and was able to erase and flash it.  Now I am having issues with Prusa Connect seeing the images.  Camera shows registration and connected on Prusa Connect. Will be researching that next.Thanks for your help!DaveSent from my iPad

image.png (view on web)

johnyHV, 2024-08-05:
@K5OZ can you send me log file from micro SD card ?

K5OZ, 2024-08-06:
Miroslav,Unfortunately, I am not home now and I didn’t put a SD card in the ESP32 CAM when I programmed it.   When I get back I will insert a card and redo the programming.Thanks againDaveSent from my iPad

closed 2025-01-03 by johnyHV