prusa3d/Prusa-Firmware-ESP32-Cam #58: Very often the configurtation web page freezes

Mr3Deee, 2024-08-04:
It is not quite clear to me why, but very often the web page simply locks up on me.

1. Changing a few settings
2. Updating the image
3. Turning on streaming
4. ...

I was considering the fact that it might be a bad internet connection but the configuration page shows me: 100% reception -37dBm which seems to be great?

Additionally I have 2 cameras, one with this firmware, showing on PrusaConnect, and one using the default CAM32 app with its own configuration app. It only streams and allows to change things, but that works just fine.

johnyHV, 2024-08-05:
Hello. Can you send me log files from micro SD card ? A too strong WiFi signal can cause problems, but it shouldn't. There might be an issue with the ESP32 processor. I had one board where the new ESP32 processor was damaged.

Mr3Deee, 2024-08-05:
Hello! Many thanks for your quick reply! I'll see that I get the logs, but
cannot do bedore a week from now. Cheers

Mr3Deee, 2024-11-21:
Hello!

I am so sorry. Had to travel a lot for the company since summer and there
is barely time to do anything at the moment.
With the X-Mas time coming up, I hope that I get some time looking into
this again (assuming that my last business related travel is really done
after the 16th).

Stefan