prusa3d/Prusa-Firmware-ESP32-Cam #59: Disconnect shortly after boot with 1.1.0

swaggner, 2024-08-05:
Using a ESP32-S3-EYE and I just updated to 1.1.0 but it won't stay connected more than 1 minute before the status light rapidly flashes.  Log file below...
[Log.txt](https://github.com/user-attachments/files/16503240/Log.txt)

johnyHV, 2024-08-08:
Hello @swaggner can you send me full log, in the verbose mode ?

swaggner, 2024-08-08:
Thanks for your help @johnyHV
I sent you an email with the serial monitor and SD card logs.  The line that stands out is 
0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 202

johnyHV, 2024-08-22:
@swaggner could you please attach the logs here? Please ensure that all sensitive content, such as passwords and tokens, is removed. It will help me navigate through the issues more easily. Thank you very much!

swaggner, 2024-08-23:
Ok.  Here you go...
[log.txt](https://github.com/user-attachments/files/16733445/log.txt)

johnyHV, 2024-09-03:
@swaggner sorry for the late response. Error 202 means that the connection to the WiFi network timed out. I see that your WiFi networks use WPA3, which could be causing issues. I'm not sure if the ESP32 supports WPA3 at the moment. It's a good question, and I need to check on that.

swaggner, 2024-09-03:
Ok.  That's interesting because that network is set to WPA2, so either that setting isn't working, or the ESP cam is trying to connect with the wrong security protocol.

swaggner, 2024-10-04:
I gave up on the ESP32-S3-EYE and ordered a ESP32-S3-CAM. I’m getting same results where it gets connected to the network but disconnects and becomes non-responsive. Anything else I can try to troubleshoot?  Are there any WiFi setting on my Unifi router that I need to change to improve stability?

swaggner, 2024-11-06:
Is there anything else I can try?  Camera will boot when plugging in power (the red internal LED will be solid red) then the LED immediately start rapidly blinking.  The internal ESP access point works, but it is unable to connect to wifi.  I even pulled out an old 802.11n router.  The camera actually connected and I thought it might be a legacy issue, but the camera disconnected shortly after and does not reattempt to reconnect.  Anything in the wifi arduino sdk that calls for specific wifi settings that I'm not utilizing?

johnyHV, 2025-01-02:
@swaggner Arduino SDK don't need nothing special. Can you try again send me log files please ?

boschhd, 2025-02-01:
Hi @swaggner ,

I saw in the log that your wifi is on channel 1. I have a different board but a similar problem, maybe this workaround also works for you:
https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/issues/104#issuecomment-2629149063