prusa3d/Prusa-Firmware-ESP32-Cam #62: Struggling flashing my Esp32-CAM

fadudba, 2024-08-22:
Hi I managed to compile and upload the software to the esp32-cam. Then I get this message in the output "Leaving...
Hard resetting via RTS pin..." and nothing happens. I press the rst button on the board and in the serial monitor I get a loop of this log:

Start MCU!
Prusa ESP32-cam https://prusa3d.cz
SW Version: 1.1.0
Build: Aug 21 2024 22:23:06
----------------------------------------------------------------
Init Logs library
Start init micro-SD Card
E (517) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).
SD Card Mount Failed
Micro-SD card not found! Disable logs
0000-00-00_00-00-00 - Init system lib
0000-00-00_00-00-00 - SW Version: 1.1.0 Build: Aug 21 2024 22:23:06
0000-00-00_00-00-00 - Board name: AI Thinker ESP32-CAM
0000-00-00_00-00-00 - CPU reset reason: Brownout reset (software or hardware)
0000-00-00_00-00-00 - CPU0 reset reason: SW_RESET
0000-00-00_00-00-00 - CPU1 reset reason: EXT_CPU_RESET
0000-00-00_00-00-00 - MCU Temperature: 41.67 *C
0000-00-00_00-00-00 - Internal Total heap: 308656 B, Internal Free Heap: 274984
0000-00-00_00-00-00 - PSRAM Total heap: 4194304 B, PSRAM Free Heap: 4192124
0000-00-00_00-00-00 - Chip model: ESP32-D0WDQ6, ChipRevision: 100, Cpu Freq: 240
0000-00-00_00-00-00 - SDK Version: v5.1.4-586-gb6b4727c58-dirty, Core Version: 3.0.4
0000-00-00_00-00-00 - Flash Size: 4194304, Flash Speed 80000000, Flash Mode: 2
0000-00-00_00-00-00 - PSRAM is used.
0000-00-00_00-00-00 - malloc/new is not using SPIRAM
0000-00-00_00-00-00 - Init cfg module: 393
0000-00-00_00-00-00 - Read FirstMcuStart FLAG
0000-00-00_00-00-00 - It's not first start MCU: 15
0000-00-00_00-00-00 - web auth enable: 0
0000-00-00_00-00-00 - web auth user: admin
0000-00-00_00-00-00 - web auth pass: ******
0000-00-00_00-00-00 - Load camera CFG from EEPROM
0000-00-00_00-00-00 - Photo quality: 10
0000-00-00_00-00-00 - Framesize: 0
0000-00-00_00-00-00 - Framesize: 0
0000-00-00_00-00-00 - brightness: 0
0000-00-00_00-00-00 - contrast: 0
0000-00-00_00-00-00 - saturation: 0
0000-00-00_00-00-00 - awb: 1
0000-00-00_00-00-00 - awb_gain: 1
0000-00-00_00-00-00 - awb_mode: 0
0000-00-00_00-00-00 - aec2: 0
0000-00-00_00-00-00 - ae_level: 0
0000-00-00_00-00-00 - aec_value: 300
0000-00-00_00-00-00 - gain_ctrl: 1
0000-00-00_00-00-00 - agc_gain: 0
0000-00-00_00-00-00 - bpc: 1
0000-00-00_00-00-00 - wpc: 1
0000-00-00_00-00-00 - raw_gama: 1
0000-00-00_00-00-00 - hmirror: 0
0000-00-00_00-00-00 - vflip: 0
0000-00-00_00-00-00 - lensc: 1
0000-00-00_00-00-00 - exposure_ctrl: 1
0000-00-00_00-00-00 - Camera flash: 0
0000-00-00_00-00-00 - Camera flash time: 200
0000-00-00_00-00-00 - Camera image rotation: 1
0000-00-00_00-00-00 - Load PrusaConnect CFG from EEPROM
0000-00-00_00-00-00 - Token: *
0000-00-00_00-00-00 - Fingerprint: MTk2Nzk1MTExOTU5MTU3IDAwOjAwOjAwOjAwOjAwOjAw
0000-00-00_00-00-00 - Refresh interval: 30
0000-00-00_00-00-00 - PrusaConnect hostname: connect.prusa3d.com
0000-00-00_00-00-00 - Time lapse function status: 0
0000-00-00_00-00-00 - SSID: 
0000-00-00_00-00-00 - WiFi password: *
0000-00-00_00-00-00 - mDNS: prusa-esp32cam
0000-00-00_00-00-00 - Enable Service AP: 1
0000-00-00_00-00-00 - Network IP method: DHCP
0000-00-00_00-00-00 - Network static IP: 255.255.255.255
0000-00-00_00-00-00 - Network static mask: 255.255.255.255
0000-00-00_00-00-00 - Network static gateway: 255.255.255.255
0000-00-00_00-00-00 - Network static DNS: 255.255.255.255
0000-00-00_00-00-00 - Service AP SSID: ESP32_camera_119.59.157
0000-00-00_00-00-00 - Service AP mode enabled
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15488
load:0x40080400,len:4
load:0x40080404,len:3180
entry 0x400805b8

can you help me please?

johnyHV, 2024-08-22:
Hello @fadudba . What hardware version of the board are you using? Could you please try uploading the firmware from the latest release?

fadudba, 2024-08-22:
> Hello @fadudba . What hardware version of the board are you using? Could you please try uploading the firmware from the latest release?

Hey I have this Board (see pic attached)
[image]

Zero-Circle, 2024-08-31:
Hey man, I had no end of issues flashing my esps but when #33 was built it took away all of my issues.

ALD-Models, 2024-08-31:
Thanks. Just trying to make the process easier for people!

Jake

closed 2024-11-18 by johnyHV