prusa3d/Prusa-Firmware-ESP32-Cam #69: unable erase/flash memory

8Sam6AcidP, 2024-09-21:
Hello 
I would like to set up an ESP32-S3-EYE using the Prusa instructions 
(https://help.prusa3d.com/guide/esp-camera-for-prusa-connect_390199)

However, I am unable to delete the memory (step 11). Nothing happens after pressing the Erase button. The following message appears in the window next to it.   

[image]

I have now gone through the instructions several times and have followed the instructions. 

Does anyone know any advice? Does anyone have the same problem?   

Thank you very much and best regards 
SaM 

Kamera ESP32-S3-EYE 
Firmware 1.1.0
Flash download tools 3.9.7
Windows 10

johnyHV, 2024-11-19:
Hello @8Sam6AcidP did you switch the board to bootloader mode before pressing ERASE?

[image]

8Sam6AcidP, 2024-11-20:
Thank you for your support. 
Yes, i switch the board to bootloader mode. 

I tried again with the OSX manual on github and changed a few commands as suggested in the comments on the Prusa site. Now the camera works. However, I don't know what the mistake was.

closed 2024-11-20 by johnyHV