prusa3d/Prusa-Firmware-ESP32-Cam #72: `Response code: Photo: 500 - unknown error code`

johnboiles, 2024-10-03:
I've set this up on a Seeed Studio Xiao ESP32-S3. Everything installed fine. I've been able to connect to WiFi and the device is showing up in Prusa Connect and reporting its IP/MAC/SSID. So I'm sure it can talk to Prusa Connect.

[image]

But no image shows up, and looking at the serial logs it looks like I'm getting an HTTP 500 from Prusa Connect. A typical log from a photo upload looks like this:

```
2024-10-03_17-22-24 - WiFi status: Wifi connected. SSID: mellonIOT, BSSID: 66:22:32:1C:C0:6A, RSSI: -63 dBm, IP: 192.168.2.235, TX power: 20dBm
2024-10-03_17-22-28 - Free RAM: 200524 B, Min: 136392
2024-10-03_17-22-28 - Free PSRAM: 7972456 B, Min: 7623552
2024-10-03_17-22-28 - MCU Temperature: 76.50 *C
2024-10-03_17-22-33 - CardStatus: 0 FileStatus: 0
2024-10-03_17-22-42 - Taking photo...
2024-10-03_17-22-42 - The picture has been saved. Size: 62624 bytes, Photo resolution: 800 x 600
2024-10-03_17-22-42 - Photo OK! 0
2024-10-03_17-22-42 - Exif header OK! Len: 310
2024-10-03_17-22-42 - Start sending photo to prusaconnect
2024-10-03_17-22-42 - Sending Photo to PrusaConnect, 62914 bytes
2024-10-03_17-22-45 - Send done: 62914/62914 bytes
2024-10-03_17-22-46 - Upload done. Response code: Photo: 500 - unknown error code ,BA:Backend available
```

I've tried various resolutions to see if sending less data would help but no luck.

Perhaps this is a temporary outage on Prusa's side. I'll update here if it starts working.

Crowdedlight, 2024-10-03:
I have the same issue. Tried to set up a new camera. My ESP32-S3-Eye logs reports the same error as yours, and same behaviour in prusa connect.

kiischti, 2024-10-03:
I have exactly thesame issue. Receved the boardtoday, flashed it without any problem, it’s fully configurable through the web interface, with live stream available, but the picture don’t show up at Prusa Connect…

johnboiles, 2024-10-04:
Seems to be working today! Probably a server-side thing on Prusa Connect

closed 2024-10-04 by johnboiles