prusa3d/Prusa-Firmware-ESP32-Cam #73: Does not automatically connect to hidden SSID

johnboiles, 2024-10-03:
I get this in the logs, presumably because my IOT devices SSID is hidden:

```
0000-00-00_00-00-00 - SSID: officeIOT not found
0000-00-00_00-00-00 - Wifi unavailable. Skip connecting to WiFi: officeIOT
```

If I connect over serial and send `wificonnect;` then it connects just fine.

xstream77, 2024-10-08:
I got the same problem with my "Ai Thinker ESP32-CAM". As long my SSID is hidden the Cam doesn't connect to the wifi. When the SSID is visible the  connection is possible.

zorro2055, 2024-10-18:
Same issue for me with the MK4S.  My SSID is hidden, and the 3D printer wifi does not connect.

MarkHinton-3D, 2024-11-07:
I also have the same issue and without this being fixed makes the products unusable in my network.

MarkHinton-3D, 2024-11-20:
> Hello. I don't know why the camera has a problem connecting to the hidden network, but I'll try to simulate it at home

Thank you,

Crowdedlight, 2024-11-21:
My expectation is that as linked in the other issue, this code is the problem that we see where it fails to connect to a hidden wifi when rebooted:

wifi_mngt.cpp#L103 
```c
if (config->CheckActifeWifiCfgFlag() == true) {
    if (true == CheckAvailableWifiNetwork(WifiSsid)) {
      WiFiStaConnect();
      log->AddEvent(LogLevel_Warning, "Connecting to WiFi: " + WifiSsid);
```

My understanding of the code from a quick glance here is that on init it checks if `WifiSsid` is visible as a WiFiNetwork. But it does it by scanning available wifis. Which will not show hidden networks, because they are not shown on a scan as they don't broadcast a ssid. 

My assumption here is that `WifiSsid` is the saved wifi ssid you have set in the config. 

Something like trying to connect to the "saved config" wifi would make more sense, and if the connection fails, then it can go "wifi unavailable" and go into AP mode.

Crowdedlight, 2024-12-04:
I ended up doing this as a hotfix, as I already had a device that was flashed previously. Then jumped on the hotspot it made and asked it to connect to our hidden network. And then unplugged and replugged the power cable to it. It connected up straight away and we have had a stable connection for the past day, compared to before when it would restart itself and fail to connect after some hours. 

It is definitely not a good solution! I have not checked what/if it breaks something else in cases where it starts up trying to connect to a wifi that isn't there. But as a hotfix to get it working here and now for us, it was fine.
 
```c
if (config->CheckActifeWifiCfgFlag() == true) {
    if (true == true) {
      WiFiStaConnect();
      log->AddEvent(LogLevel_Warning, "Connecting to WiFi: " + WifiSsid);
    }
```

johnyHV, 2025-01-02:
Hello @Crowdedlight . It's true. This solution works for hide network.