prusa3d/Prusa-Firmware-ESP32-Cam #74: Snapshot fails to uplaod to PrusaConnect - 500 error in log

ArtSpaziano, 2024-10-04:
Snapshot and stream look fine on prusa-esp32cam.local but snapshots fail to upload to PrusaConnect.
Using 1.1.0
[FullESP32-Cam-log.txt](https://github.com/user-attachments/files/17252165/FullESP32-Cam-log.txt)

Snippet of log entry:

024-10-04_01-59-32 - MCU Temperature: 56.11 *C
2024-10-04_01-59-35 - WiFi status: Wifi connected. SSID: Spazianos_2G, BSSID: A0:36:BC:82:5E:20, RSSI: -61 dBm, IP: 192.168.1.163, TX power: 19.5dBm
2024-10-04_01-59-35 - Taking photo...
2024-10-04_01-59-35 - The picture has been saved. Size: 2868 bytes, Photo resolution: 320 x 240
2024-10-04_01-59-35 - Photo OK! 0
2024-10-04_01-59-35 - Exif header OK! Len: 310
2024-10-04_01-59-35 - Start sending photo to prusaconnect
2024-10-04_01-59-35 - Sending Photo to PrusaConnect, 3158 bytes
2024-10-04_01-59-36 - Send done: 3158/3158 bytes
2024-10-04_01-59-37 - Upload done. Response code: Photo: 500 - unknown error code ,BA:Backend available
2024-10-04_01-59-37 - Save TimeLaps photo to SD card
2024-10-04_01-59-37 - Photo saved to SD card. EXIF

Plan on getting another ESP32 Cam to install 1.0.3 shortly to see if the problem goes away but it would be good to get this working, since it's the latest.

feyded1020, 2024-10-04:
Currently having the same issue, just started happening earlier today. Have tried to readd the camera, but still receive error 500.

ArtSpaziano, 2024-10-04:
It works this morning. It looks like bad timing on my part because I chose to put the camera into service on the same day Prusa was having problems with their server.
All well now - closing this.

closed 2024-10-04 by ArtSpaziano