prusa3d/Prusa-Firmware-ESP32-Cam #75: RHYX-M21-45 camera boot issues

tomu4, 2024-10-04:
I got this ESP32-S3-CAM from Aliexpress as a bundle with RHYX-M21-45 camera (according to the description, it should have been OV2640, but that's not the point). I tried loading compiled firmware, which caused bootloop. I tried compiling my own from source files, and got camera initialization error 106, as I found online, it's 0x106 - Operation or feature not supported. I therefore changed   CameraConfig.pixel_format in camera.cpp from   PIXFORMAT_JPEG to PIXFORMAT_RGB565. That solved the bootloop issue, I am able to load the configuration website with no problem, but it doesn't show an image, neither is the image uploaded to Prusa Connect. I suppose it's because camera's lack of JPEG support. Is there more troubleshooting I could try, or do I just have to buy original OV2640 camera with JPEG support?

alfs, 2024-10-06:
The RHYX labeled camera seems to be a GC2145, not OV2640.
Also you may need to limit resolution to QVGA or QQVGA.

tomu4, 2024-10-09:
> The RHYX labeled camera seems to be a GC2145, not OV2640.
Yeah, I figured it's not OV2640 (despite labeled as so). I just wanted to know whether this particular camera was even supported. 
> Also you may need to limit resolution to QVGA or QQVGA.
You mean in addition to changing format, or keep format set to JPEG and limit resolution?

alfs, 2024-10-12:
> Yeah, I figured it's not OV2640 (despite labeled as so). I just wanted to know whether this particular camera was even supported.
>
>
> You mean in addition to changing format, or keep format set to [...]

It doesn't support compression (so no JPEG at all), meaning that higher resolutions may require too long transmission time. You can of course try higher resolutions - just something to try/consider if it isn't working.

tomu4, 2024-10-12:
Well, I solved it by buying OV2640 instead. Even if I got it to work, the quality would suck anyway. I just think there should be updated documentation for this particular camera type, since it's exactly this one that's on photos in github documentation, and if it doesn't work out of the box, there should at least be information on how to make it work. Anyway, for me, the problem is solved, so I'm going to close the issue, but I believe someone else is for sure going to have the same problem.

spyder12342, 2024-12-19:
[image]
I am using qv260 but i am having the same issue

Robotiox, 2025-02-22:
I also got that problem. Please can someone help me?

gtelzur, 2025-02-23:
Me too... I decided to order the one that is known to work, i.e. OV2640,  and not a poor substitute

jayraj2827, 2025-04-15:
But what do if someone has that camera module someone have Any Solution for this??

manojcode4, 2025-05-04:
Bro I also face some error but I try in rgb565 image save in sd card but not support to display

tomu4, 2025-08-29:
> but if change that camera, and i use OV2640 instead of using GC2145 will it work ??

Yes, OV2640 works as intended 🙂