prusa3d/Prusa-Firmware-ESP32-Cam #78: “The snapshot is not available” in Prusa Connect

ZevEisenberg, 2024-10-22:
I have an ESP32-CAM running the firmware binaries from version 1.1.0 of this repo. I have my Connect Token from Prusa Connect entered. If I visit http://prusa-esp32cam.local/action_capture locally, I get a 200. I'm seeing images come in from the camera. However, in Prusa Connect, I see "The snapshot is not available." If I inspect the Prusa Connect web page and look at the console, I see errors like this:

>[Error] Failed to load resource: the server responded with a status of 404 () (last, line 0)

And the body of the response is:

```json
{
    "message": "Snapshot not found",
    "code": "NOT_FOUND_SNAPSHOT"
}
```

The readme mentions:

>If you are getting a 400 return code from Prusa Connect, then you need to create a new other camera in Prusa Connect and enter a new token into the ESP32 camera.

But a) I'm getting a 404, not a 400, and b) I already tried deleting and recreating a new camera with a new token several times.

Any suggestions? Would logs help? I'll have to dig up a microSD card to write them, if so, but it seems like the upload is working? 🤔

Maintraxx, 2024-10-30:
Is your printer online? In my case, the snapshot is only available, when the printer itself is online.

dzwiedziu-nkg, 2024-10-30:
Confirm. I just configured [this](https://www.printables.com/model/736656-xlcam-prusa-xl-raspberry-pi-camera) and works only when printer is online.

ZevEisenberg, 2024-10-31:
Thank you! That worked Am I right in thinking that the camera is connecting directly to the printer, rather than posting to Prusa Connect via a web API? If so, it might be nice to update the docs to indicate this: https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/pull/83

closed 2024-11-20 by johnyHV