prusa3d/Prusa-Firmware-ESP32-Cam #79: Where to buy in Europe?

bodokaiser, 2024-10-23:
Hi all,

thanks for sharing and extending this beautiful project!

For the last days, I have been searching the internet for a distributor where I can buy any of the supported camera modules. However, I have failed to find the exact same version in stock and deliverable to the EU (Germany). I have tried Amazon, Conrad, Reichelt, RS-components, Mouser, and Digikey.

It would be great to add links of potential distributors to the supported boards section in the README.md or at least share some links unofficially in this thread.

mozgy, 2024-10-23:
Mouser sends all it's 'stuff' from Italy warehouse.

bodokaiser, 2024-10-23:
Thank you!

> https://www.az-delivery.de/en/products/esp32-cam-modul-esp32-wifi-bluetooth-modul-inklusive-kamera

Which firmware would I have to use for this one?

Maintraxx, 2024-10-30:
You also find it on amazon with USB Port, which is very helpful: https://www.amazon.de/gp/product/B0D9BSRYF9

Then you can install the firmware online with a browser: https://ald-models.github.io/Prusa-ESP32-Cam-Web-Upload/

closed 2024-11-19 by johnyHV