prusa3d/Prusa-Firmware-ESP32-Cam #81: Network hidden

MarkHinton-3D, 2024-10-29:
I have a network with an SSID that is hidden. When I make the network available as invisible the camera connects fine.

With exactly the same credentials in place when I re-hide the network it disappears and will not reconnect.

Maintraxx, 2024-10-30:
I have the same issue!

stho64, 2025-02-04:
Hi Johny,

I have the same here and think I see the problem in 
wifi_mngt.cpp

line 103:

102   if (config->CheckActifeWifiCfgFlag() == true) {
103    **if (true == CheckAvailableWifiNetwork(WifiSsid)) {**
104      WiFiStaConnect();

Your function: CheckAvailableWifiNetwork() ==> ScanWifiNetwork(i_data) check the SSID is available but a hidden WIFI is not shown (available == false) and so WiFiStaConnect(); is not called on a boot/reboot.

I played with different ESP32 Cams: WROVER, S3 WROOM and ESP32 S3 Cam. All have with same problem. 

Via serial terminal I can connect to my hidden Wifi with using the commands:
setwifissid | Setting WiFi SSID, where variable SSID is network name
setwifipass | Setting WiFi password, where variable PASSWORD is WiFi password
wificonnect | Connecting to WiFi network - **THIS WORKS WELL**

After a reboot it won't connect to WIFI as it should, but a serial entered wificonnect connects again for a while.

I'm not really familiar with Arduino IDE / compiling so I hope you can fix this issue with my details. May be a checkbox for "hidden Wifi"  in the Wifi config area which skips the function CheckAvailableWifiNetwork() without ScanWifiNetwork for the finding the best one.

Thanks in advance, Steffen