prusa3d/Prusa-Firmware-ESP32-Cam #84: No picture visible on PRUSA Connect... WHY?

HydroLuz, 2024-11-13:
Hello everybody,

I have carefully followed the instructions for the Freenove ESP32-WROVER and succesfully connected it to PRUSA Connect. Snapshots are constantly being send to PRUSA Connect, but I cannot see anything..
[image]
Does anyone know why?

I have tested the camera with the Arduino sketch provided by Freenove and it works. However, with on PRUSA Connect I only see a black image...

Any solutions? I would be very grateful :)

johnyHV, 2024-11-19:
Hello @HydroLuz . Do you still have the same problem?

HydroLuz, 2024-11-19:
> Hello @HydroLuz . Do you still have the same problem?

@johnyHV unfortunately I do... I have tried out the arduino sketch aswell, however I also just see black images even though it appears to send pictures to Prusa Connect..
Now I have tested the arduino sketch provided by Freenove again and it shows me footage (see picture). I really understand what is the problem...

[image]

HydroLuz, 2024-11-19:
> > Hello @HydroLuz . Do you still have the same problem?
> 
> @johnyHV unfortunately I do... I have tried out the arduino sketch aswell, however I also just see black images even though it appears to send pictures to Prusa Connect.. Now I have tested the arduino sketch provided by Freenove again and it shows me footage (see picture). I really understand what is the problem...
> 
> ![WhatsApp Image 2024-11-19 at 14 10 16](https://private-user-images.githubusercontent.com/188344623/387668251-3501feef-e168-4844-82ea-1d0643f8bbe7.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMjIzNDcsIm5iZiI6MTczMjAyMjA0NywicGF0aCI6Ii8xODgzNDQ2MjMvMzg3NjY4MjUxLTM1MDFmZWVmLWUxNjgtNDg0NC04MmVhLTFkMDY0M2Y4YmJlNy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTlUMTMxNDA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjI1ZDA3YmEzNmNmNjI1YzFmNmZlNjViNDY4MDk0YmNiZjAxM2M5MzIxOWYxMzZiODJiMGE0ZmZhYTEyOWIyNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.iIUjoDcS9yu-DKmkIHb-zYmqS2msYkrEsB1kjRdt8eA)

Here you can see that the camera is registered with Prusa Connect: 
[image]

HydroLuz, 2024-11-19:
> Hello @HydroLuz . Do you still have the same problem?

Here you can see the Wifi connection status:
[image]
 and here the System status:
[image]

johnyHV, 2024-11-20:
@HydroLuz Strange. Please, can you send me board picture ? From both side. Thanks

HydroLuz, 2024-11-21:
> @HydroLuz Strange. Please, can you send me board picture ? From both side. Thanks

[image]
[image]

nhuhoai, 2024-11-29:
I have the same issue: I can compile and upload into my microchip. Everything works except the camera: black image.

I uploaded the ESP32 Camera Web Server from the official example of the board and the camera is working. I had checked the pinout but it's identical and check the log in verbose mode but everything seems ok.

After some tests, I change the camera settings in the prusa esp32 cam web server:

1. Set the `brightness` to `high`
   [image]
2. Enable the `second level automatic exposure control` in `advanced settings`
   [image]

And voilà, I have a picture now. Hope you have the same issue and it can help you.

HydroLuz, 2024-12-02:
> I have the same issue: I can compile and upload into my microchip. Everything works except the camera: black image.
>
> I uploaded the ESP32 Camera Web Server from the official example of the board and [...]

HEYYYYY! You are right! IT WORKED! Thank you very much!
Interersting fact: After I made the changes you recommended and I finally got a picture I changed the resolution and image quality and it turned black again. I was furious as resetting did not bring the picture back. However, after I disabled the second level automatic exposure control in advanced settings the picture returned hahahahahahahhahaha.

Thanky again my friedn :)

nhuhoai, 2024-12-02:
You're welcome. But I have the same issue, after while, I got the black screen again, just edit some camera settings the picture returned. But, we can make it. It seems a software problem with the automatic settings.

HydroLuz, 2024-12-03:
> You're welcome. But I have the same issue, after while, I got the black screen again, just edit some camera settings the picture returned. But, we can make it. It seems a software problem with the automatic settings.

Unfortunately, I experienced the same. After a while I got the black screen again aswell. Not sure why...

nhuhoai, 2024-12-03:
I have the "black picture" after each reboot. But this time, my camera is falling and it's face to the LEDs. It captures this picture:

[image]

Meaning the black screen is related to the camera settings on boot.

I make some tests with the UI:

1. Adding onload script on the picture
2. Analyze the picture (light level), if too dark, then wait for the camera settings (from /json_input) and get the contrast value (A)
3. Force update contrast setting with a different value (with /set_int?contrast=X)
4. Force update contrast setting with the current value A (with /set_int?contrast=A)
5. Reload image adding a timestamp param in the URL

It works when I load the web interface but I need to go on the web interface to fix the problem. The fix should be in the microchip code.

HydroLuz, 2024-12-04:
> I have the "black picture" after each reboot. But this time, my camera is falling and it's face to the LEDs. It captures this picture: [...]

Does it work constantly or does it go back to black again after a while?
Btw. do you also use the Freenove ESP32 Wrover?

nhuhoai, 2024-12-04:
> Does it work constantly or does it go back to black again after a while? Btw. do you also use the Freenove ESP32 Wrover?

After manual fix, it works until the next reboot (the max time use is about 4 hours).

About the microchip, I use an ESP32-S3-CAM.

HydroLuz, 2024-12-09:
[image]
Super weird! After days with a black image suddenly we got a "clear" image again without me changing anything hahahahahaha... I don´t really get it xD

drunkrhin0, 2025-09-13:
> I have the same issue: I can compile and upload into my microchip. Everything works except the camera: black image.
>
> I uploaded the ESP32 Camera Web Server from the official example of the board and [...]

Thank you @nhuhoai