prusa3d/Prusa-Firmware-ESP32-Cam #85: ESP32-CAM not working

Stoupa77, 2024-11-15:
Hello
I bought two ESP32-CAM cameras, and I set them up according to the instructions for connecting to PrusaConnect and the cameras are running. When I connect to their wi-fi it works, when I add it to my network it runs. I can also see it on http://prusa-esp32cam.local. I add it to PrusaConnect. The camera appears there. It shows me what wi-fi it is connected to, what IP it has, but it does not send an image (the image is not available) - instead there is only a crossed out image of the camera. It does not show either one, it always stays at the same point. The camera update is checked.
As soon as I turn off the camera and return to it the next day, it logs in to my wi-fi, but I can no longer access it via the PC, it is no longer displayed at all. But it sends information to PrusaConnect. It shows the IP address, but I cannot connect to the IP of the camera. Only to the camera's wi-fi and then through it, but nothing else.
Please can you advise me where the problem is, why it doesn't work?

Thank you for the advice?

johnyHV, 2024-11-19:
Hello @Stoupa77 . Have you tried turning on the camera and letting it run for, for example, 90 seconds, without trying to connect to the camera via WiFi?

Or can you insert to the camera micro SD card, and then send me a log file? You can add logs here to git. Passwords are not displayed

Stoupa77, 2024-11-21:
Hello @johnyHV 

Yes, I tried it for a longer time, when I didn't even touch the camera, but the effect was the same.
I'll try putting a card in it when I get to it again and send the log.

Stoupa77, 2024-11-24:
Hello.

I am attaching the log info in the attachment. I had to delete the camera again, flash it and set a new token on prusa connet. Then it failed because I accidentally clicked on authorization and had to flash it again and put the token again. Now it only works on the IP address and I can only access it there. In Prusa Connet it reports itself as registered, but nothing is displayed. There is only a crossed out image of the camera (picture). Now I will let it run, for example for an hour. Then I will turn it off and tomorrow I will not be able to access it even on the same IP address as it has today.

PRUSA CONNECT IMAGE
[image]

LOG INFO
[Log info.txt](https://github.com/user-attachments/files/17894943/Log.info.txt)

johnyHV, 2024-11-26:
It's strange. And on the camera WEB interface can you see photo ? Can you try using the latest FW version? I've just released it, it's FW 1.1.1. I don't think it will help. But we can try it.

Stoupa77, 2024-11-27:
I did the update, but nothing changed, as you wrote. It loads the photo normally on the page, but nothing is sent to Prusa Connect. The camera has been running for about 4 days without turning off and is not sending anything anymore.

Stoupa77, 2024-12-01:
I left the camera turned off for two days and when I turned it on, I can't connect to it anymore. It won't connect to Wi-Fi, I can't connect to its Wi-Fi. It's on and the light is blinking that it's connected to Wi-Fi, but it's not there.

peterwup, 2024-12-12:
Same problem here. 
Local picture available in the Web  page, but not in PrusaConnect
[image]

I have my ESp32 CAM connected by USB and the following LOG file, where I can see no error:

```

12:32:51.012 -> ----------------------------------------------------------------
12:32:51.012 -> Init Logs library
12:32:51.012 -> Start init micro-SD Card
12:32:51.045 -> E (76) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
12:32:51.045 -> E (76) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
12:32:51.078 -> SD Card Mount Failed
12:32:51.078 -> Micro-SD card not found! Disable logs
12:32:51.078 -> 0000-00-00_00-00-00 - Init system lib
12:32:51.078 -> 0000-00-00_00-00-00 - SW Version: 1.1.0 Build: Dec 12 2024 09:45:31
12:32:51.078 -> 0000-00-00_00-00-00 - Board name: AI Thinker ESP32-CAM
12:32:51.078 -> 0000-00-00_00-00-00 - CPU reset reason: Reset due to power-on event
12:32:51.078 -> 0000-00-00_00-00-00 - CPU0 reset reason: POWERON_RESET
12:32:51.078 -> 0000-00-00_00-00-00 - CPU1 reset reason: EXT_CPU_RESET
12:32:51.111 -> 0000-00-00_00-00-00 - MCU Temperature: 63.33 *C
12:32:51.111 -> 0000-00-00_00-00-00 - Internal Total heap: 308852 B, Internal Free Heap: 274964
12:32:51.111 -> 0000-00-00_00-00-00 - PSRAM Total heap: 4194304 B, PSRAM Free Heap: 4192124
12:32:51.111 -> 0000-00-00_00-00-00 - Chip model: ESP32-D0WDQ6, ChipRevision: 101, Cpu Freq: 240
12:32:51.111 -> 0000-00-00_00-00-00 - SDK Version: v5.1.4-972-g632e0c2a9f-dirty, Core Version: 3.0.7
12:32:51.144 -> 0000-00-00_00-00-00 - Flash Size: 4194304, Flash Speed 80000000, Flash Mode: 0
12:32:51.145 -> 0000-00-00_00-00-00 - PSRAM is used.
12:32:51.145 -> 0000-00-00_00-00-00 - malloc/new is not using SPIRAM
12:32:51.145 -> 0000-00-00_00-00-00 - Init cfg module: 393
12:32:51.145 -> 0000-00-00_00-00-00 - Read FirstMcuStart FLAG
12:32:51.145 -> 0000-00-00_00-00-00 - It's not first start MCU: 15
12:32:51.145 -> 0000-00-00_00-00-00 - web auth enable: 0
12:32:51.145 -> 0000-00-00_00-00-00 - web auth user: admin
12:32:51.177 -> 0000-00-00_00-00-00 - web auth pass: ******
12:32:51.177 -> 0000-00-00_00-00-00 - Load camera CFG from EEPROM
12:32:51.177 -> 0000-00-00_00-00-00 - Photo quality: 10
12:32:51.177 -> 0000-00-00_00-00-00 - Framesize: 6
12:32:51.177 -> 0000-00-00_00-00-00 - Framesize: 6
12:32:51.177 -> 0000-00-00_00-00-00 - brightness: 0
12:32:51.177 -> 0000-00-00_00-00-00 - contrast: 0
12:32:51.177 -> 0000-00-00_00-00-00 - saturation: 0
12:32:51.177 -> 0000-00-00_00-00-00 - awb: 1
12:32:51.211 -> 0000-00-00_00-00-00 - awb_gain: 1
12:32:51.211 -> 0000-00-00_00-00-00 - awb_mode: 0
12:32:51.211 -> 0000-00-00_00-00-00 - aec2: 1
12:32:51.211 -> 0000-00-00_00-00-00 - ae_level: 0
12:32:51.211 -> 0000-00-00_00-00-00 - aec_value: 300
12:32:51.211 -> 0000-00-00_00-00-00 - gain_ctrl: 1
12:32:51.211 -> 0000-00-00_00-00-00 - agc_gain: 0
12:32:51.211 -> 0000-00-00_00-00-00 - bpc: 1
12:32:51.211 -> 0000-00-00_00-00-00 - wpc: 1
12:32:51.211 -> 0000-00-00_00-00-00 - raw_gama: 1
12:32:51.211 -> 0000-00-00_00-00-00 - hmirror: 0
12:32:51.211 -> 0000-00-00_00-00-00 - vflip: 0
12:32:51.243 -> 0000-00-00_00-00-00 - lensc: 1
12:32:51.243 -> 0000-00-00_00-00-00 - exposure_ctrl: 1
12:32:51.243 -> 0000-00-00_00-00-00 - Camera flash: 0
12:32:51.243 -> 0000-00-00_00-00-00 - Camera flash time: 200
12:32:51.243 -> 0000-00-00_00-00-00 - Camera image rotation: 1
12:32:51.243 -> 0000-00-00_00-00-00 - Load PrusaConnect CFG from EEPROM
12:32:51.243 -> 0000-00-00_00-00-00 - Token: *********************
12:32:51.243 -> 0000-00-00_00-00-00 - Fingerprint: ***************************************
12:32:51.277 -> 0000-00-00_00-00-00 - Refresh interval: 30
12:32:51.277 -> 0000-00-00_00-00-00 - PrusaConnect hostname: connect.prusa3d.com
12:32:51.277 -> 0000-00-00_00-00-00 - Time lapse function status: 0
12:32:51.277 -> 0000-00-00_00-00-00 - SSID: MySSID
12:32:51.277 -> 0000-00-00_00-00-00 - WiFi password: *****************
12:32:51.277 -> 0000-00-00_00-00-00 - mDNS: prusa-esp32cam
12:32:51.277 -> 0000-00-00_00-00-00 - Enable Service AP: 0
[... 122 lines cut ...]
12:34:25.287 -> 2024-12-12_11-34-23 - Free RAM: 120844 B, Min: 47044
12:34:25.349 -> 2024-12-12_11-34-23 - Free PSRAM: 3808036 B, Min: 3772772
12:34:25.349 -> 2024-12-12_11-34-23 - MCU Temperature: 65.00 *C
12:34:27.512 -> 2024-12-12_11-34-25 - Taking photo...
12:34:27.692 -> 2024-12-12_11-34-25 - The picture has been saved. Size: 81178 bytes, Photo resolution: 1600 x 1200
12:34:27.728 -> 2024-12-12_11-34-25 - Photo OK! 0
12:34:27.728 -> 2024-12-12_11-34-25 - Exif header OK! Len: 310
12:34:27.728 -> 2024-12-12_11-34-25 - Start sending photo to prusaconnect
12:34:27.728 -> 2024-12-12_11-34-25 - Sending Photo to PrusaConnect, 81468 bytes
12:34:28.843 -> 2024-12-12_11-34-26 - Send done: 81468/81468 bytes
12:34:29.865 -> 2024-12-12_11-34-27 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
12:34:35.279 -> 2024-12-12_11-34-33 - CardStatus: 0 FileStatus: 0
12:34:47.355 -> 2024-12-12_11-34-45 - WiFi status: Wifi connected. SSID: MySSID, BSSID: E8:DF:70:68:36:A5, RSSI: -65 dBm, IP: 192.168.100.91, TX power: 19.5dBm
12:34:55.287 -> 2024-12-12_11-34-53 - Free RAM: 120844 B, Min: 47044
12:34:55.333 -> 2024-12-12_11-34-53 - Free PSRAM: 3808036 B, Min: 3772772
12:34:55.333 -> 2024-12-12_11-34-53 - MCU Temperature: 64.44 *C
12:34:58.440 -> 2024-12-12_11-34-56 - Taking photo...
12:34:58.623 -> 2024-12-12_11-34-56 - The picture has been saved. Size: 80481 bytes, Photo resolution: 1600 x 1200
12:34:58.657 -> 2024-12-12_11-34-56 - Photo OK! 0
12:34:58.657 -> 2024-12-12_11-34-56 - Exif header OK! Len: 310
12:34:58.657 -> 2024-12-12_11-34-56 - Start sending photo to prusaconnect
12:34:58.657 -> 2024-12-12_11-34-56 - Sending Photo to PrusaConnect, 80771 bytes
12:34:59.761 -> 2024-12-12_11-34-57 - Send done: 80771/80771 bytes

```

closed 2025-01-02 by johnyHV