prusa3d/Prusa-Firmware-ESP32-Cam #97: Frequent disconnect from wifi

zahumenskyduri, 2024-12-21:
hello,

i have got new AI Thinker ESP32-CAM//After successfull flash adn configuration the camera connects to my wifi, but disconnects quite often. I have Unifi APs. No other device has problem just this one.

i see lot of **WiFi disconnected from access point. Reason: 8** 

Start MCU!
SW Version: 1.1.0 ,Build: Dec 19 2024 20:38:23
Verbose mode: false
Log level: 1
0000-00-00_00-00-00 - CPU reset reason: Software reset via esp_restart
0000-00-00_00-00-00 - CPU0 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - CPU1 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - Service AP mode disabled!
0000-00-00_00-00-00 - Connecting to WiFi: Sv@Wifi
2024-12-21_12-44-52 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_12-45-38 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_12-45-38 - WiFi disconnected from access point. Reason: 8
2024-12-21_12-49-00 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_12-49-22 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_12-49-22 - WiFi disconnected from access point. Reason: 8
2024-12-21_12-50-46 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_12-50-46 - WiFi disconnected from access point. Reason: 8
2024-12-21_12-56-45 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_12-57-18 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_12-57-18 - WiFi disconnected from access point. Reason: 8
2024-12-21_12-58-49 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_12-59-38 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_12-59-38 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-01-24 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-01-58 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-01-58 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-11-44 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-12-14 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-12-14 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-17-57 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-18-18 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-18-18 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-24-08 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-24-50 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-24-50 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-26-13 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-26-42 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-26-42 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-34-59 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-35-34 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-35-34 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-37-03 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-37-26 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-37-26 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-40-40 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-41-10 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-41-10 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-49-27 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-50-02 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-50-02 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-51-02 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-51-26 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-51-26 - WiFi disconnected from access point. Reason: 8
2024-12-21_13-58-14 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_13-58-54 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_13-58-54 - WiFi disconnected from access point. Reason: 8
2024-12-21_14-03-55 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_14-04-30 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_14-04-30 - WiFi disconnected from access point. Reason: 8
2024-12-21_14-13-14 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_14-13-50 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_14-13-50 - WiFi disconnected from access point. Reason: 8
2024-12-21_14-16-50 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_14-17-34 - Reconnecting to WiFi. STA. Problem with connecting to backend!
[... 85 lines cut ...]
2024-12-21_16-42-27 - Disconnect from WiFi
2024-12-21_16-42-27 - Reconnecting to WiFi. STA
2024-12-21_16-42-29 - WiFi disconnected from access point. Reason: 15
2024-12-21_16-42-32 - WiFi disconnected from access point. Reason: 15
2024-12-21_16-42-35 - WiFi disconnected from access point. Reason: 15
2024-12-21_16-42-38 - WiFi disconnected from access point. Reason: 15
2024-12-21_16-42-41 - WiFi disconnected from access point. Reason: 15
2024-12-21_16-42-43 - WiFi WDG. STA connection lost. 1
2024-12-21_16-42-43 - WiFi STA connection lost. WDG timer expired. Restart MCU!
2024-12-21_16-42-43 - WiFi disconnected from access point. Reason: 8
----------------------------------------------------------------
Start MCU!
SW Version: 1.1.0 ,Build: Dec 19 2024 20:38:23
Verbose mode: false
Log level: 1
0000-00-00_00-00-00 - CPU reset reason: Software reset via esp_restart
0000-00-00_00-00-00 - CPU0 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - CPU1 reset reason: SW_CPU_RESET
0000-00-00_00-00-00 - Service AP mode disabled!
0000-00-00_00-00-00 - Connecting to WiFi: Sv@Wifi
2024-12-21_16-46-32 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_16-46-32 - WiFi disconnected from access point. Reason: 8
2024-12-21_16-48-28 - ERROR SEND DATA TO SERVER! INCORRECT DATA LENGTH!
2024-12-21_16-48-52 - Reconnecting to WiFi. STA. Problem with connecting to backend!
2024-12-21_16-48-52 - WiFi disconnected from access point. Reason: 8

johnyHV, 2025-01-02:
Hello @zahumenskyduri . What is the quality of the wifi signal?

zahumenskyduri, 2025-01-02:
was ok, AP is in same room. will test it and get back with exact number

zahumenskyduri, 2025-01-03:
Connection status
Status: Connected
SSID: Sv@Wifi
Signal: 100% / -37dBm

logs (warning level)
[log.txt](https://github.com/user-attachments/files/18303579/log.txt)

johnyHV, 2025-01-03:
@zahumenskyduri Thank you for the log file. Please, can you try flash latest FW version 1.1.2 ? 

All reason codes are stored here https://github.com/espressif/arduino-esp32/blob/04963009eedfbc1e0ea2e1378ae69e7cebda6fd6/tools/sdk/include/esp32/esp_event_legacy.h 

```    
    SYSTEM_EVENT_WIFI_READY = 0,           /**< ESP32 WiFi ready */
    SYSTEM_EVENT_SCAN_DONE,                /**< ESP32 finish scanning AP */
    SYSTEM_EVENT_STA_START,                /**< ESP32 station start */
    SYSTEM_EVENT_STA_STOP,                 /**< ESP32 station stop */
    SYSTEM_EVENT_STA_CONNECTED,            /**< ESP32 station connected to AP */
    SYSTEM_EVENT_STA_DISCONNECTED,         /**< ESP32 station disconnected from AP */
    SYSTEM_EVENT_STA_AUTHMODE_CHANGE,      /**< the auth mode of AP connected by ESP32 station changed */
    SYSTEM_EVENT_STA_GOT_IP,               /**< ESP32 station got IP from connected AP */
    SYSTEM_EVENT_STA_LOST_IP,              /**< ESP32 station lost IP and the IP is reset to 0 */
    SYSTEM_EVENT_STA_WPS_ER_SUCCESS,       /**< ESP32 station wps succeeds in enrollee mode */
    SYSTEM_EVENT_STA_WPS_ER_FAILED,        /**< ESP32 station wps fails in enrollee mode */
    SYSTEM_EVENT_STA_WPS_ER_TIMEOUT,       /**< ESP32 station wps timeout in enrollee mode */
    SYSTEM_EVENT_STA_WPS_ER_PIN,           /**< ESP32 station wps pin code in enrollee mode */
    SYSTEM_EVENT_AP_START,                 /**< ESP32 soft-AP start */
    SYSTEM_EVENT_AP_STOP,                  /**< ESP32 soft-AP stop */
    SYSTEM_EVENT_AP_STACONNECTED,          /**< a station connected to ESP32 soft-AP */
    SYSTEM_EVENT_AP_STADISCONNECTED,       /**< a station disconnected from ESP32 soft-AP */
    SYSTEM_EVENT_AP_STAIPASSIGNED,         /**< ESP32 soft-AP assign an IP to a connected station */
    SYSTEM_EVENT_AP_PROBEREQRECVED,        /**< Receive probe request packet in soft-AP interface */
    SYSTEM_EVENT_GOT_IP6,                  /**< ESP32 station or ap or ethernet interface v6IP addr is preferred */
    SYSTEM_EVENT_ETH_START,                /**< ESP32 ethernet start */
    SYSTEM_EVENT_ETH_STOP,                 /**< ESP32 ethernet stop */
    SYSTEM_EVENT_ETH_CONNECTED,            /**< ESP32 ethernet phy link up */
    SYSTEM_EVENT_ETH_DISCONNECTED,         /**< ESP32 ethernet phy link down */
    SYSTEM_EVENT_ETH_GOT_IP,               /**< ESP32 ethernet got IP from connected AP */
    SYSTEM_EVENT_MAX
    ```
    
    Reason 8 is  SYSTEM_EVENT_STA_LOST_IP,              /**< ESP32 station lost IP and the IP is reset to 0 */

zahumenskyduri, 2025-01-03:
did du update, but still happend. made verbose log
[log.txt](https://github.com/user-attachments/files/18304466/log.txt)

zahumenskyduri, 2025-01-12:
so disconnect got resolved with 1.1.2..but when i want to stream it crashes the camera